hadess/iio-sensor-proxy #107: net.hadess.SensorProxy was not provided by any .service files

dinojr, 2016-10-26:
I'm running Gnome 3.22 in Ubuntu 16.10 on a Microsoft Surface Pro. The auto-rotate icon has disappeared from the Gnome User Menu since the last upgrade. I'm trying to debug the problem using the guidelines you provide:

udevadm info --export-db gives several iio entries (see attached file)

hadess, 2016-10-26:
Looks likely that you didn't install it in the right place. Or `iio-sensor-proxy` could not detect those particular sensors.

Check the output of `systemctl status iio-sensor-proxy.service` and attach the output here.

dinojr, 2016-10-26:
Actually I just noticed the service was simply not started: simply running
> sudo systemctl start iio-sensor-proxy.service
gets screen rotation working.

What's the proper way to have the service started at startup ?

hadess, 2016-10-26:
It's started automatically through udev. You shouldn't need to do anything to get it started, other than maybe rebooting the machine.

dinojr, 2016-10-27:
Turns out the iio-sensor-proxy package installs the file 80-iio-sensor-proxy.rules in /rules.d/, not /lib/udev/rules.d/. It looks as a bug to me, should I report it upstream ?

I added a link to this file in /lib/udev/rules.d/ and the service is now launched at startup. However it doesn't work: monitor-sensor reports that the orientation is not detected. If however I remove the link (so that udev doesn't start it) and if I run it with systemctl, it's working fine.

hadess, 2016-10-27:
> Turns out the iio-sensor-proxy package installs the file 80-iio-sensor-proxy.rules in /rules.d/, not  lib/udev/rules.d/. It looks as a bug to me, should I report it upstream ?

You mean downstream I guess. You didn't have the udev development libraries installed when it was built, so `$PKG_CONFIG --variable=udevdir udev` was empty, hence the installation. I've fixed this now.

> I added a link to this file in /lib/udev/rules.d/ and the service is now launched at startup. However it doesn't work: monitor-sensor reports that the orientation is not detected. If however I remove the link (so that udev doesn't start it) and if I run it with systemctl, it's working fine.

See #82

closed 2016-10-27 by hadess

dinojr, 2016-10-28:
Thanks,
`Type=idle`
instead of
`Type=dbus`
seems to have fixed it

commit 78b4656f (2016-10-27): build: Check for .pc files before using them

Otherwise we might install the udev rules in /rules.d.

Closes #107