hadess/iio-sensor-proxy #108: How to fix Wisky tablet screen orientation off by 90 cw?

oh2emd, 2016-10-28:
After grub screen turns 180 degrees and after login screen rotates always off 90 degrees cw. If I disable iio-sensor-proxy -service I can use tablet, but how to fix autorotation permanently?

udevadm info --export-db

`P: /devices/platform/80860F41:04/i2c-12/i2c-INT33F4:00/axp288_adc/iio:device1`
`N: iio:device1`
`E: DEVNAME=/dev/iio:device1`
`E: DEVPATH=/devices/platform/80860F41:04/i2c-12/i2c-INT33F4:00/axp288_adc/iio:device1`
`E: DEVTYPE=iio_device`
`E: MAJOR=245`
`E: MINOR=1`
`E: SUBSYSTEM=iio`
`E: SYSTEMD_WANTS=iio-sensor-proxy.service`
`E: TAGS=:systemd:`
`E: USEC_INITIALIZED=12775906`

hadess, 2016-10-28:
That sounds like:
https://bugs.freedesktop.org/show_bug.cgi?id=94894
which you'll need to fix before we can think of having a quirk for this device.

oh2emd, 2016-10-28:
I put **GRUB_CMDLINE_LINUX="i915.fastboot=2"** to grub and now orientation is normal after grub. It killed autorotation, so screen does not autorotate anymore.

hadess, 2016-10-28:
Right, you'll want to report this on the bug, so it can be reopened.

hadess, 2016-11-22:
This is a bug in the i915 kernel driver which doesn't use the boot information to set the initial state of the screen. We can't/won't add a quirk for it in iio-sensor-proxy.

closed 2016-11-22 by hadess