hadess/iio-sensor-proxy #110: Lenovo ThinkPad Yoga 260

sdahdah, 2016-11-06:
It seems iio-sensor-proxy is not able to detect this device's sensors (light, accelerometer).

`monitor-sensor` displays: `Waiting for iio-sensor-proxy to appear` indefinitely.

Let me know what info you'd need and I can provide it.

michauds, 2016-11-09:
The sensors won't be detected since the kernel isn't yet aware of them. Here's the article on the relevant patch: https://lkml.org/lkml/2016/5/31/3

FYI, the above mentioned patch is included in kernel 4.9. 

I'll ping back once I've had more time to test this out.

zauguin, 2016-11-13:
I use a ThinkPad Yoga 460 (it uses the ISH too) with kernel 4.9. The light sensor works, but the accelerometer only works with `drv-iio-poll-accel.c` and not with `drv-iio-buffer-accel.c`.

I don't know anything about the iio subsystem, but if you need to test some code I'm happy to help.

hadess, 2016-11-22:
> so a small patch was required.

What patch?

zauguin, 2016-11-22:
zauguin@97e523f1edb227c4fd030a3bb205882d623f3579

hadess, 2016-11-22:
This shouldn't be needed. Test with the current master please.

zauguin, 2016-11-22:
You're right. It works with the current master. Awesome!

hadess, 2016-11-22:
Great :)

closed 2016-11-22 by hadess