hadess/iio-sensor-proxy #112: Lenovo Yoga 910

henrythor, 2016-11-11:
Hi!

I'm having trouble getting the sensors working on my Yoga 910. I'm running Antergos, iio-sensor-proxy version 1.3

Here's my debug output. Anyone have any pointers?

```
** (process:2933): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device0
** (process:2933): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device0 of type ambient light sensor at IIO Buffer Light sensor
** (process:2933): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device1
** (process:2933): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device1 of type accelerometer at IIO Buffer accelerometer
** (process:2933): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/trigger1
** (process:2933): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2933): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2933): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2933): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2933): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2933): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2933): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/trigger0
** (process:2933): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2933): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2933): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2933): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2933): DEBUG: No new data available on 'iio:device0'
** (process:2933): DEBUG: No new data available on 'iio:device1'
** (process:2933): DEBUG: No new data available on 'iio:device0'
** (process:2933): DEBUG: No new data available on 'iio:device1'
** (process:2933): DEBUG: No new data available on 'iio:device0'
** (process:2933): DEBUG: No new data available on 'iio:device1'
** (process:2933): DEBUG: No new data available on 'iio:device0'
** (process:2933): DEBUG: No new data available on 'iio:device1'
```

Monitor says:
```
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
```

I checked journalctl, and found this:
```
Nov 11 19:22:53 tempest kernel: i2c_hid i2c-ITE8186:00: error in i2c_hid_init_report size:19 / ret_size:18
```

Does that mean I have a driver issue?

charkins, 2016-11-20:
I am seeing the same behavior on Yoga 900 on Fedora 25 w/ kernel 2.8.7, including the same i2c_hid error, despite Yoga 900 having been reported as working with iio-sensor-proxy previously. iio-sensor-proxy lists orientation as 'undefined'  and does not update, despite the data changing in sysfs in_accel_*_raw. monitor-sensor is reporting changes from abient light sensor.

hadess, 2016-11-22:
Worked-around in master, see #116 for details.

closed 2016-11-22 by hadess

henrythor, 2016-11-22:
I cloned the repo and tested. I still have some issues.

At first the light sensor jumped from 0 to 2.0 lux, but I wasn't able to get it to change with a flashlight.
```
➜  ~ monitor-sensor -v
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
    Light changed: 2.000000 (lux)
^C
```

Accelerometer still shows undefined orientation.

I ran it manually with debug messages and got this:
```
** (process:2419): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1
** (process:2419): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1 of type ambient light sensor at IIO Buffer Light sensor
** (process:2419): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0
** (process:2419): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:2419): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/trigger0
** (process:2419): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/scan_elements/in_accel_y_en
** (process:2419): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/scan_elements/in_accel_x_en
** (process:2419): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/scan_elements/in_accel_z_en
** (process:2419): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2419): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2419): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2419): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2419): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2419): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:2419): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/trigger1
** (process:2419): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1/scan_elements/in_intensity_both_en
** (process:2419): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2419): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2419): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2419): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2419): DEBUG: No new data available on 'iio:device1'
** (process:2419): DEBUG: No new data available on 'iio:device0'
** (process:2419): DEBUG: No new data available on 'iio:device1'
** (process:2419): DEBUG: No new data available on 'iio:device0'
** (process:2419): DEBUG: No new data available on 'iio:device1'
** (process:2419): DEBUG: No new data available on 'iio:device0'
** (process:2419): DEBUG: No new data available on 'iio:device1'
** (process:2419): DEBUG: No new data available on 'iio:device0'
```

hadess, 2016-11-22:
Please file a separate bug, and instead of relaunching the daemon, edit the `.service` file to enable debug, and reboot the machine. Attach the output of `journalctl -b -u iio-sensor-proxy.service`.