hadess/iio-sensor-proxy #113: Build fails on ARM

dteleguin, 2016-11-13:
We at Mageia Linux are working on an ARM port. Here is log excerpt for iio-sensor-proxy-1.3:

```
iio-buffer-utils.c: In function 'process_scan_1':
iio-buffer-utils.c:495:20: error: cast increases required alignment of target type [-Werror=cast-align]
     guint32 val = *(guint32 *) (data + buffer_data->channels[k]->location);
                    ^
iio-buffer-utils.c:501:19: error: cast increases required alignment of target type [-Werror=cast-align]
     gint32 val = *(gint32 *) (data + buffer_data->channels[k]->location);
                   ^
cc1: all warnings being treated as errors
```

Full logs: [ARMv5](http://pkgsubmit.mageia.org/uploads/failure/cauldron/core/release/20161113160324.akien.duvel.24652/log/botcmd.1479053164.armlet1.log), [ARMv7](http://pkgsubmit.mageia.org/uploads/failure/cauldron/core/release/20161113160324.akien.duvel.24652/log/botcmd.1479053164.armlet1.log). Please let me know if the logs expire.

dteleguin, 2016-11-13:
As you might have noticed, Mageia has rather strict policy on compiler warnings. There was another similar issue:
```
fake-input-accelerometer.c: In function 'send_uinput_event':
fake-input-accelerometer.c:90:2: error: ignoring return value of 'write', declared with attribute warn_unused_result [-Werror=unused-result]
  write (data->uinput, &ev, sizeof(ev));
  ^
fake-input-accelerometer.c:94:2: error: ignoring return value of 'write', declared with attribute warn_unused_result [-Werror=unused-result]
  write (data->uinput, &ev, sizeof(ev));
  ^
fake-input-accelerometer.c:98:2: error: ignoring return value of 'write', declared with attribute warn_unused_result [-Werror=unused-result]
  write (data->uinput, &ev, sizeof(ev));
  ^
fake-input-accelerometer.c:104:2: error: ignoring return value of 'write', declared with attribute warn_unused_result [-Werror=unused-result]
  write (data->uinput, &ev, sizeof(ev));
  ^
cc1: all warnings being treated as errors
```

As a workaround, I've used `export WARN_CFLAGS="-Wno-unused-result -Wno-error=unused-result"` before `configure`. But with the current issue I'm afraid there's a risk of producing a non-working package by simply suppressing these warnings.

hadess, 2016-11-22:
Got a patch?

parkerlreed, 2017-08-16:
Same issue is happening on Arch with GCC 7.1.1

`export WARN_CFLAGS="-Wno-unused-result -Wno-error=unused-result"` did let it compile. (Waiting on kernel build to test)

commit ebc348ab (2017-08-16): iio-buffer-utils: Fix cast-align warnings on ARM

iio-buffer-utils.c: In function 'process_scan_1':
iio-buffer-utils.c:495:20: error: cast increases required alignment of target type [-Werror=cast-align]
     guint32 val = *(guint32 *) (data + buffer_data->channels[k]->location);
                    ^
iio-buffer-utils.c:501:19: error: cast increases required alignment of target type [-Werror=cast-align]
     gint32 val = *(gint32 *) (data + buffer_data->channels[k]->location);
                   ^

Looks like there's no need to cast.

Closes: #113
closed 2017-08-16 by hadess