hadess/iio-sensor-proxy #116: Remove artificial startup delay

hadess, 2016-11-22:
As per https://github.com/hadess/iio-sensor-proxy/commit/5468452f7a72566d2e6ddc44f9396d3d088fa9fb

We need to debug the kernel and find the root cause of the problem. Polling IIO drivers such as the discrete accelerometers in cheaper tablets have no problems starting up without a delay, so the problem is likely in the hid-sensor-hub code.

rickysarraf, 2016-12-21:
With newer kernels, it is getting worse. I'm not sure if anyone on the Linux side is even working on finding the actual bug. Like others, I found the rotation broken yesterday, on my 4.9 kernel.

I'm back on 4.8.15 with iio-sensor-proxy 2.0, and everything is working fine.

hadess, 2016-12-28:
> With newer kernels, it is getting worse. I'm not sure if anyone on the Linux side is even working on finding the actual bug. Like others, I found the rotation broken yesterday, on my 4.9 kernel.

Unrelated bug. This is the fix:
https://www.spinics.net/lists/linux-input/msg48540.html

rickysarraf, 2016-12-28:
Yes. This was mentioned in here too: https://github.com/hadess/iio-sensor-proxy/issues/124#issuecomment-269447602
But I just built the kernel, and it doesn't seem to make much a difference.

virtualguywithabowtie, 2017-01-09:
has this been confirmed to fix in surface pro 3?

westernmagic, 2017-01-11:
You could move it to the systemd service file instead - that way it can be customized by the user.

```
[Unit]
Description=IIO Sensor Proxy service

[Service]
Type=dbus
BusName=net.hadess.SensorProxy
TimeoutStartSec=15
ExecStartPre=/usr/bin/sleep 5
ExecStart=/usr/bin/iio-sensor-proxy
#Uncomment this to enable debug
#Environment="G_MESSAGES_DEBUG=all"

# Lockdown
ProtectSystem=true
ProtectControlGroups=true
ProtectHome=true
ProtectKernelModules=true
PrivateTmp=true
PrivateNetwork=true
MemoryDenyWriteExecute=true
RestrictRealtime=true
```

hadess, 2017-01-11:
I'm not interested in making it configurable, the end goal is fixing the kernel. I don't have hardware on which I can bisect the problem, so until I get that hardware, or somebody else root causes the problem, there's really nothing to do.

smithfred, 2017-06-14:
> has this been confirmed to fix in surface pro 3?

No, rotation is still broken on the SP3, at least on Fedora 25/iio-sensor-proxy 2.2.1/kernel 4.11.3; see my comment on #58.

rickysarraf, 2017-06-26:
For folks on 4.9 and 4.11 kernels, there was a regression introduced in the stable tree, which was later fixed. Not sure which stable minor versions they were part of.

`iio: hid-sensor: Store restore poll and hysteresis on S3`
`commit 5d9854eaea776441b38a9a45b4e6879524c4f48c upstream`

With this fix, rotation has been reliable so far.

hadess, 2017-06-27:
> With this fix, rotation has been reliable so far.

My tests were with 4.12.0-rc3, which contains the fix[1], and you'll still need the delay to work-around the actual bug in the kernel.

[1]: 
```
git describe 5d9854eaea776441b38a9a45b4e6879524c4f48c
v4.11-rc4-8-g5d9854eaea77
```

hadess, 2017-08-11:
See possible fix referenced in https://github.com/hadess/iio-sensor-proxy/issues/82#issuecomment-321699993

closed 2017-08-16 by hadess

keeferrourke, 2017-08-18:
I'm running Fedora 26 on a Lenovo Yoga 710 and still have this problem, despite the fix committed earlier. Orientation is still not detected unless the system is first suspended, or the iio-sensor-proxy service is forced to restart some time after booting (after logging in for instance).

```
$ uname -a
Linux audino 4.12.5-300.fc26.x86_64 #1 SMP Mon Aug 7 15:27:25 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```
```
$ cat /etc/os-release 
NAME=Fedora
VERSION="26 (Workstation Edition)"
ID=fedora
VERSION_ID=26
PRETTY_NAME="Fedora 26 (Workstation Edition)"
ANSI_COLOR="0;34"
CPE_NAME="cpe:/o:fedoraproject:fedora:26"
HOME_URL="https://fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=26
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=26
PRIVACY_POLICY_URL=https://fedoraproject.org/wiki/Legal:PrivacyPolicy
VARIANT="Workstation Edition"
VARIANT_ID=workstation
```
It seems the race condition is still present; this machine boots very quickly (~1.5s). Suspending after boot seems to initialize the drivers properly, as is commonly mentioned.

hadess, 2017-08-18:
This version doesn't have the fix. The README clearly says that 4.13 is the first version with the fix.

keeferrourke, 2017-08-18:
@hadess my bad, sorry. >.<

hadess, 2017-08-23:
> @hadess my bad, sorry. >.< earlier in this thread, it's mentioned that 4.12 had it. Cyclical bug references had me chasing documentation.

Will be fixed in the v4.12.8 Fedora 26 kernel if you want to test. Please file any problems you see with that version under a new bug.

smithfred, 2017-10-08:
As of 4.13.4-200.fc26 rotation is still broken on the Surface Pro 3.

This issue is referenced from #58 as being a workaround, so from my reading of this ticket, rotation should now be working on the above kernel if that were the case.

As it stands though, the same workaround is still needed (`watch /sys/bus/iio/devices/iio:device*/in_accel_x_raw`).

Can you re-open #58 on that basis unless I'm misunderstanding something?

hadess, 2017-10-08:
I already mentioned to file a new bug. My guess is that the Surface Pro 3 needs quirks in the kernel to spit out data, unrelated to this problem.

smithfred, 2017-10-20:
You mentioned to another commenter on this bug to open another bug, and he was commenting on completely different hardware. I replied on this bug because you closed #58 with "Worked-around in master, see #116 for details."

The issue reported in #58 was never fixed; can you please re-open it, since it already exists.

sarangnemo, 2017-10-21:
@smithfred Kernel 4.14 finally fixes the problem on my Surface Pro 3!

smithfred, 2017-10-22:
@sarangnemo what distro? For me with the latest 4.14rc5 build on Fedora:

Good news: don't need to  continually poll a /sys file for automatic rotation to occur
Bad news: now any small movement results in 2-3 rapid random rotations, with the final rotation bearing no relation to the physical orientation :|

sarangnemo, 2017-10-22:
@smithfred It's Manjaro (based on Arch). But see #195 -- the most recent commit to iio-sensor-proxy introduced a different issue which sounds like might lead to your problem. Downgrading to the previous commit or even v2.2 will hopefully give you the expected behavior as it does for me.

smithfred, 2017-10-24:
@sarangnemo yep I've just confirmed that downgrading to 2.2.x results in working auto-rotation (with 4.14), hooray! 😃 Thanks for the pointer.

commit 20c41c11 (2017-08-16): Revert "main: Add artificial delay before starting daemon"

This reverts commit 5468452f7a72566d2e6ddc44f9396d3d088fa9fb as
the problem was root-caused and fixed in the upstream kernel.

Please see README for the fix' details.

Closes: #116