hadess/iio-sensor-proxy #117: Lenovo Ideapad Yoga 910

henrythor, 2016-11-22:
I'm having difficulties getting sensors working on my Yoga 910.

Here's journalctl output with debugging enabled:
```
Nov 22 18:43:03 tempest systemd[1]: Starting IIO Sensor Proxy service...
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Found als at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Found device /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1 of type ambient light sensor at IIO Buffer Light sensor
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Found accel_3d at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Found device /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/trigger0
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/scan_elements/in_accel_y_en
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/scan_elements/in_accel_x_en
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/scan_elements/in_accel_z_en
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/trigger1
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1/scan_elements/in_intensity_both_en
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
Nov 22 18:43:06 tempest iio-sensor-prox[295]: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
Nov 22 18:43:06 tempest systemd[1]: Started IIO Sensor Proxy service.
Nov 22 18:43:15 tempest iio-sensor-prox[295]: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
Nov 22 18:43:16 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:17 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:17 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:18 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:19 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:20 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:20 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:21 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:22 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:22 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:23 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:23 tempest iio-sensor-prox[295]: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
Nov 22 18:43:23 tempest iio-sensor-prox[295]: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
Nov 22 18:43:24 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:24 tempest iio-sensor-prox[295]: No new data available on 'iio:device1'
Nov 22 18:43:24 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:25 tempest iio-sensor-prox[295]: No new data available on 'iio:device1'
Nov 22 18:43:25 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:25 tempest iio-sensor-prox[295]: No new data available on 'iio:device1'
Nov 22 18:43:26 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:26 tempest iio-sensor-prox[295]: No new data available on 'iio:device1'
Nov 22 18:43:27 tempest iio-sensor-prox[295]: No new data available on 'iio:device0'
Nov 22 18:43:27 tempest iio-sensor-prox[295]: No new data available on 'iio:device1'
```

Here's the gdbus introspect:
```
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};
```

And here's udevadm's output http://pastebin.com/Q9QA0A7G

hadess, 2016-11-23:
Does increasing the sleep in `src/iio-sensor-proxy.c` and rebooting fix the problem?

henrythor, 2016-11-23:
I tried 6 seconds and 10 seconds. Did not notice a change other than that now it says the orientation is 'normal' instead of 'undefined', but it won't update.

```
-- Logs begin at Tue 2016-11-08 11:09:49 GMT, end at Wed 2016-11-23 20:16:08 GMT. --
Nov 23 20:15:00 tempest systemd[1]: Starting IIO Sensor Proxy service...
Nov 23 20:15:10 tempest iio-sensor-prox[325]: Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1
Nov 23 20:15:10 tempest iio-sensor-prox[325]: Found device /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1 of type ambient light sensor at IIO Buffer Light sensor
Nov 23 20:15:10 tempest iio-sensor-prox[325]: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0
Nov 23 20:15:10 tempest iio-sensor-prox[325]: Found device /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
Nov 23 20:15:10 tempest iio-sensor-prox[325]: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/trigger0
Nov 23 20:15:10 tempest iio-sensor-prox[325]: Enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/scan_elements/in_accel_y_en
Nov 23 20:15:10 tempest iio-sensor-prox[325]: Enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/scan_elements/in_accel_x_en
Nov 23 20:15:10 tempest iio-sensor-prox[325]: Enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/scan_elements/in_accel_z_en
Nov 23 20:15:11 tempest iio-sensor-prox[325]: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 23 20:15:11 tempest iio-sensor-prox[325]: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 23 20:15:11 tempest iio-sensor-prox[325]: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 23 20:15:11 tempest iio-sensor-prox[325]: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 23 20:15:11 tempest iio-sensor-prox[325]: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 23 20:15:11 tempest iio-sensor-prox[325]: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
Nov 23 20:15:11 tempest iio-sensor-prox[325]: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/trigger1
Nov 23 20:15:11 tempest iio-sensor-prox[325]: Enabled sensor /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1/scan_elements/in_intensity_both_en
Nov 23 20:15:14 tempest iio-sensor-prox[325]: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
Nov 23 20:15:14 tempest iio-sensor-prox[325]: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
Nov 23 20:15:14 tempest systemd[1]: Started IIO Sensor Proxy service.
Nov 23 20:16:01 tempest iio-sensor-prox[325]: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
Nov 23 20:16:02 tempest iio-sensor-prox[325]: process_scan_1: channel_index: 0, chan_name: in_accel_x, channel_data_index: 0 location: 0
Nov 23 20:16:02 tempest iio-sensor-prox[325]: process_scan_1: channel_index: 1, chan_name: in_accel_y, channel_data_index: 1 location: 4
Nov 23 20:16:02 tempest iio-sensor-prox[325]: process_scan_1: channel_index: 2, chan_name: in_accel_z, channel_data_index: 2 location: 8
Nov 23 20:16:02 tempest iio-sensor-prox[325]: Accel read from IIO on 'iio:device0': 62, -1039, -167 (scale 0.009807)
Nov 23 20:16:02 tempest iio-sensor-prox[325]: Accel sent by driver (quirk applied): -62, 1039, -167 (scale: 0.009807)
Nov 23 20:16:02 tempest iio-sensor-prox[325]: Emitted orientation changed: from undefined to normal
Nov 23 20:16:02 tempest iio-sensor-prox[325]: No new data available on 'iio:device0'
Nov 23 20:16:03 tempest iio-sensor-prox[325]: No new data available on 'iio:device0'
Nov 23 20:16:04 tempest iio-sensor-prox[325]: No new data available on 'iio:device0'
Nov 23 20:16:05 tempest iio-sensor-prox[325]: No new data available on 'iio:device0'
Nov 23 20:16:05 tempest iio-sensor-prox[325]: No new data available on 'iio:device0'
Nov 23 20:16:06 tempest iio-sensor-prox[325]: No new data available on 'iio:device0'
Nov 23 20:16:08 tempest iio-sensor-prox[325]: No new data available on 'iio:device0'
```

hadess, 2016-11-23:
Does it work when suspending and then resuming the machine?

henrythor, 2016-11-23:
No, that doesn't appear to have any effect. You mean suspend-to-memory, not disk, yes?

henrythor, 2016-11-23:
What about this line (from earlier issue). It's still there, do you think it's related?

```
Nov 23 20:15:01 tempest kernel: i2c_hid i2c-ITE8186:00: error in i2c_hid_init_report size:19 / ret_size:18
```

hadess, 2016-11-24:
Yeah, that probably doesn't help. I'm running out of ideas as to what the problem could be I'm afraid.

henrythor, 2016-11-24:
Here's what the driver reads from the accelerometer. These values don't change no matter how much I wave my laptop around like an idjit.

(from /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-8/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/)

```
in_accel_offset: 0
in_accel_sampling_frequency: 8.000000
in_accel_scale: 0.009806650
in_accel_x_raw: 62
in_accel_y_raw: 64497
in_accel_z_raw: 65369
```

aeonsablaze, 2017-03-23:
I'm seeing this issue as well. Also on a Yoga 910, but running Fedora 25. Anything I can do to help?

My readings, for what its worth:
```
in_accel_hysteresis: 0.020000
in_accel_offset: 0
in_accel_sampling_frequency: 50.000000
in_accel_scale: 0.009806650
in_accel_x_raw: 65529
in_accel_y_raw: 960
in_accel_z_raw: 65408
```

Would love to get this solved. Currently if I don't keep hid_sensor_accel_3d blacklisted my screen is permanently upside down (and figuring that out with an upside down screen and no other computer available wasn't exactly a picnic). :(

seanvk, 2017-05-13:
Fedora 25
Yoga 910

Same issue.  To work around it I have to blacklist hid_sensor_accel_3d.

Sean

CTXz, 2017-07-14:
This is likely to be an issue regarding the Yoga 910's HID Sensor HUB driver (and as stated, hence falls under driver issues).

In the past, a couple of Lenovo devices (the yoga 900 included) suffered from possibly the same issue, which was caused due to a miss matching HID HUB ID. The issue has been resolved by explicitly defining device specific IDs for HID sensor hub over i2c communication, called quirks. Those quirks are enumerated under /drivers/hid/hid-ids.h. I'm going to email the authors from those previous patches, and ask how exactly they have obtained those quirk IDs.

References :
https://patchwork.kernel.org/patch/7329051/ - Patch for yoga 2
https://patchwork.kernel.org/patch/7887361/ - Patch for yoga 900

CTXz, 2017-07-15:
I've managed to get the ITE 8186 sensor hub working by adding it's appropriate quirk. Both screen rotation and automatic brightness is now working perfectly fine on my Lenovo 910, with the exception of the suspend bug (see **Known problems** in this projects readme file).

A patch has been send and will hopefully be merged by the next kernel release

https://patchwork.kernel.org/patch/9842347/

hadess, 2017-08-16:
From https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#known-problems:

> Every Linux kernel from 4.3 up to version 4.12 had a bug that made
> made iio-sensor-proxy fail to see any events coming from sensors until the 
> sensor was power-cycled (unplugged and replugged, or suspended and resumed).
> 
> The bug was finally fixed in [this commit](https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/commit/?h=fixes-togreg&id=eafad73ed3851707fa6e3124a255fc049ff9545d)
> in the upstream kernel and backported to stable releases. If you experience
> unresponsive sensors, ask your distributor to make sure this patch was 
> applied to the version you're using.

Please reopen if you still have problems with a kernel containing this fix.

closed 2017-08-16 by hadess