hadess/iio-sensor-proxy #118: Sensors not detected on HP Spectre x360 (Kaby Lake)

Pointedstick, 2016-11-23:
In Fedora 25 with GNOME 3.22, I get nothing:

```
$ udevadm info --export-db | grep iio_device | wc -l
0

# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:4351): DEBUG: Could not find any supported sensors
```

I know the sensors are present since in Windows 10, the screen correctly rotates and the OS receives signals to disable the keyboard and run in tablet mode and do things like that.

hadess, 2016-11-23:
I'm afraid that there's nothing that iio-sensor-proxy can do to fix the lack of drivers in the kernel. You might want to file a bug against Fedora to enable the kernel driver, if there is one, or send a mail to the linux-iio list if there isn't, so that they can help with finding information about it.

closed 2016-11-23 by hadess

Pointedstick, 2016-11-23:
Thanks for the info. I have filed https://bugzilla.redhat.com/show_bug.cgi?id=1398002

Pointedstick, 2016-12-29:
FYI in Kernel 4.9 or later, the sensors are detected just fine. So you can add my machine to the "tested on" list.

hadess, 2017-01-02:
Done, thanks.