hadess/iio-sensor-proxy #119: Light level never changes ASUS UX32VD

artemjackson, 2016-11-26:
I don't actually know where should I ask about the issue, so asked here.

The value of light level is never changed. It's constantly 50.000000 (unit: lux) even if sensor is fully covered.
```
$ systemctl stop iio-sensor-proxy.service
$ sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:1824): DEBUG: Found IIO poll light at /sys/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0008:00/iio:device0
** (process:1824): DEBUG: Found device /sys/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0008:00/iio:device0 of type ambient light sensor at IIO Polling Light sensor
** (process:1824): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:1824): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:1824): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:1824): DEBUG: Light level sent by driver (quirk applied): 50.000000 (unit: lux)
** (process:1824): DEBUG: Emitted light changed: from 0.000000 to 50.000000
** (process:1824): DEBUG: Light level sent by driver (quirk applied): 50.000000 (unit: lux)
** (process:1824): DEBUG: Light level sent by driver (quirk applied): 50.000000 (unit: lux)
...
** (process:1824): DEBUG: Light level sent by driver (quirk applied): 50.000000 (unit: lux)
** (process:1824): DEBUG: Light level sent by driver (quirk applied): 50.000000 (unit: lux)
** (process:1824): DEBUG: Light level sent by driver (quirk applied): 50.000000 (unit: lux)
...
...and so on
```
Definitely something works wrong. I have some assumptions: either kernel als driver works wrong or iio-sensor-prox reads from a wrong place.

Tested on
```
Asus UX32Vd
Linux 4.8.10-1-ARCH #1 x86_64
iio-sensor-proxy 1.3-1
Gnome 3.22.2
```

olerem, 2017-03-29:
ASUS ACPI is providing actual interface for acpi_als *and* a switch controller over special ASUS WMI interface to provide mesured data.
If switch is not enabled, ACPI will provide only some static value.
asus-nb-wmi and asus-wmi modules are responsible for triggering this switch.

hadess, 2017-08-16:
From https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#known-problems:

> Every Linux kernel from 4.3 up to version 4.12 had a bug that made
> made iio-sensor-proxy fail to see any events coming from sensors until the 
> sensor was power-cycled (unplugged and replugged, or suspended and resumed).
> 
> The bug was finally fixed in [this commit](https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/commit/?h=fixes-togreg&id=eafad73ed3851707fa6e3124a255fc049ff9545d)
> in the upstream kernel and backported to stable releases. If you experience
> unresponsive sensors, ask your distributor to make sure this patch was 
> applied to the version you're using.

Please reopen if you still have problems with a kernel containing this fix.

closed 2017-08-16 by hadess

olerem, 2018-09-22:
This functionality on ASUS laptops should be explicitly enabled, it is disabled by default. I was the one who provided kernel patches to make it work.
Currently it is working on Ubuntu 18.04 with kernel 4.15.. on ASUS UX330UAK.
In this notebook model ASUS made a Bug - which I reported to ASUS and got usual answer: we don't care, Linux is not supported and so on.
With this bug, you can't read sensor switch status any more, so you should only blindly enable or disable it. The status function is trying to execute variable instead of a function, which make it even worse. Windows drivers seems not to use this part of functionality...
@gdevenyi in you case, the sensor seems to be disabled. So, you will need to find why.