hadess/iio-sensor-proxy #120: lenovo carbon x1 issues with rotation

matthiasclasen, 2016-11-30:
I am seeing a rotation lock icon in the shell status menu, but the screen does not actually rotate.
The control-center doesn't offer me manual rotation either. This is an X session, and using xrandr -o works. iio-sensor-proxy appears to be running.

Output of udevadm --export-db is attached.
[udev-dump.txt](https://github.com/hadess/iio-sensor-proxy/files/622876/udev-dump.txt)

hadess, 2016-11-30:
This looks very much like the acer-wmi driver is creating an accelerometer device on a Lenovo laptop, as in: http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/acer-wmi.c#n1878

It's possible we never noticed before because iio-sensor-proxy just wasn't there to try to use it.

Could you please attach a `journalctl -b` output after a clean boot? If it shows:
`Acer Laptop ACPI-WMI Extras`
in the output, then we have a culprit.

hadess, 2016-11-30:
See also issue #71.

matthiasclasen, 2016-12-01:
> If it shows:
> Acer Laptop ACPI-WMI Extras
> in the output, then we have a culprit.

Yes, it does.
[boot.txt](https://github.com/hadess/iio-sensor-proxy/files/624214/boot.txt)

hadess, 2016-12-05:
The acer-wmi's matching is broken:
http://www.spinics.net/lists/platform-driver-x86/index.html#09227
and this shouldn't be loaded on this machine.

hadess, 2016-12-05:
Closing as that should be fixed in the kernel.

closed 2016-12-05 by hadess