hadess/iio-sensor-proxy #121: Fix trigger name for non HID sensor devices

hadess, 2016-12-01:
In `src/drv-iio-buffer-accel.c`, in `get_trigger_name()` we look for triggers named `accel_3d-dev%s`. This won't work if that's not the name of the associated trigger.

Tofee, 2017-04-21:
I confirm this issue on the Cube i9 tablet, where the name is "i2c-KIOX000A:00".
However I don't understand why this code browses through all the iio subsystem devices, when it does have the device itself under the hand. Why not simply replace the whole loop with
'trigger_name = g_udev_device_get_sysfs_attr (device, "name")' ?

hadess, 2017-04-26:
> Why not simply replace the whole loop with 'trigger_name = g_udev_device_get_sysfs_attr (device, "name")' ?

That's the trigger associated with the device, once it's been setup. It's what we're trying to *set*, so we obviously can't read it from there. On some devices/drivers combinations, the default will already be setup, but we need to be able to set it, and detect it in a robust way. I don't have access to hardware that can use this, so if you figure out a way...

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/121

closed 2020-01-09 by hadess