hadess/iio-sensor-proxy #122: lis3lv02d accelerometer no longer detected

phomes, 2016-12-02:
Since 531dcc80f6dc190d813056625a5e93d03ff0f801 my accelerometer is no longer detected by iio-sensor-proxy. Reverting that commit makes it see it again.

hadess, 2016-12-03:
Did you install all the files in the correct location?

hadess, 2016-12-03:
Works as expected with `fake-input-accelerometer`. This is the output of `udevadm monitor -p` when starting the fake device:
```
UDEV  [1399.004848] add      /devices/virtual/input/input30/event12 (input)
ACTION=add
DEVNAME=/dev/input/event12
DEVPATH=/devices/virtual/input/input30/event12
ID_INPUT=1
ID_INPUT_ACCELEROMETER=1
IIO_SENSOR_PROXY_TYPE=input-accel
MAJOR=13
MINOR=76
SEQNUM=2955
SUBSYSTEM=input
SYSTEMD_WANTS=iio-sensor-proxy.service
TAGS=:systemd:
USEC_INITIALIZED=1399004392
```

If you have the SYSTEMD_WANTS of `iio-sensor-proxy.service` but not the driver type in the list, then it's an installation problem.

phomes, 2016-12-04:
Yes, sorry. Installation problem. Basically fedora 25 has this in 80-iio-sensor-proxy.rules:
# iio-sensor-proxy
# IIO sensor to D-Bus proxy

SUBSYSTEM=="iio", ACTION=="add", TAG+="systemd", ENV{SYSTEMD_WANTS}+="iio-sensor-proxy.service"
SUBSYSTEM=="input", ACTION=="add", TAG+="systemd", ENV{ID_INPUT_ACCELEROMETER}=="1", ENV{SYSTEMD_WANTS}+="iio-sensor-proxy.service"
SUBSYSTEM=="hwmon", ACTION=="add", TAG+="systemd", ENV{SYSTEMD_WANTS}+="iio-sensor-proxy.service"

I now dumped the one shipped here into /etc/udev/rules.d and now I have IIO_SENSOR_PROXY_TYPE=input-accel

closed 2016-12-04 by phomes