hadess/iio-sensor-proxy #123: Not Working on Acer Switch Alpha 12

ragibkl, 2016-12-07:
This laptop is not in your list of tested laptops. Can you help support this laptop as well?
Laptop Model : Acer Switch Alpha 12
Distro : Fedora 25
`Linux alpha12 4.9.0-0.rc7.git3.2.fc26.x86_64 #1 SMP Thu Dec 1 22:50:14 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

Some shell output:
[shell_iio_proxy.txt](https://github.com/hadess/iio-sensor-proxy/files/636890/shell_iio_proxy.txt)

Thank you.

ragibkl, 2016-12-07:
Update:
My issue seems similar to the following reddit post: https://www.reddit.com/r/SurfaceLinux/comments/3sk5tp/autorotation_issues_plus_a_workaround/
related: https://github.com/hadess/iio-sensor-proxy/issues/58
workaround: `watch -n 0.1 cat '/sys/bus/iio/devices/iio:device'*/*raw*`
with the workaround, screen orientation and light sensor seems to work.

harriseldon, 2016-12-14:
@ragibkl 

I also have a switch alpha 12 and noticed that iio sensors are not working either, but for a different reason. For me, it cannot open /dev/input/event9 although it runs as root. I am on arch and I will check there. I am looking forward to the driver bug fix.

I am willing to install the git version to test it out.

jin-eld, 2016-12-25:
Hi there,

I'm running (or at least trying to run) Fedora 25 on an Acer Switch 10, I have a similar problem like @harriseldon.

The log shows an endless loop of:
Dec 25 13:55:39 localhost.localdomain iio-sensor-prox[722]: Could not open input accel '/dev/input/event8': Operation not permitted

iio-sensor-proxy is running as root

ragibkl, 2017-01-18:
@hadess , I have some updates.
I am now running Korora 25 Linux (a Fedora 25 remix), with latest updates.
Linux 4.9.3-200.fc25.x86_64 #1 SMP
package version: iio-sensor-proxy 2.0-1.fc25

Everything works now. Like magic. No workarounds. ootb functional (after running updates).
So, maybe bug was fixed in kernel update?

Thanks for all the hard work. I appreciate it.

jin-eld, 2017-01-18:
Hmm, interesting, I have the same kernel on Fedora 25, also pulled in via the latest updates, but it did not change anything for me, I still get the same error :(

corpix, 2017-01-25:
Just for information:
There is a ticket in [RedHat bugzilla](https://bugzilla.redhat.com/show_bug.cgi?id=1408858).

harriseldon, 2017-01-27:
I just upgraded to kernel 4.9.6 and I no longer get the cannot open input device error. I can also see changes to ambient light using monitor-sensor. I did not try orientation yet.

jin-eld, 2017-02-01:
Updated to kernel-4.9.6-200.fc25.x86_64, unfortunately no change for me, still the same behaviour.

IdleGandalf, 2017-02-02:
Got the same problem on an Acer Switch Alpha 12 with Arch's 4.9.6 kernel, the accelerometer is just not accessible.

ragibkl, 2017-05-05:
Update: Hi @hadess , after several system updates, this stopped working for me.
I'm on kernel 4.10.13-200.fc25.x86_64
So, perhaps this is really kernel issue?
My experience is similar to #159

FFortissimo, 2017-05-05:
The workaround doesn't work for me:
```
[jeroen@dds ~]$ watch -n 0.1 cat '/sys/bus/iio/devices/iio:device'*/*raw*
watch: failed to parse argument: '0.1'
```

hadess, 2017-08-16:
From https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#known-problems:

> Every Linux kernel from 4.3 up to version 4.12 had a bug that made
> made iio-sensor-proxy fail to see any events coming from sensors until the 
> sensor was power-cycled (unplugged and replugged, or suspended and resumed).
> 
> The bug was finally fixed in [this commit](https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/commit/?h=fixes-togreg&id=eafad73ed3851707fa6e3124a255fc049ff9545d)
> in the upstream kernel and backported to stable releases. If you experience
> unresponsive sensors, ask your distributor to make sure this patch was 
> applied to the version you're using.

Please reopen if you still have problems with a kernel containing this fix.

closed 2017-08-16 by hadess