hadess/iio-sensor-proxy #124: Startup on X1 Yoga still seems to be unable to read IIO data

hellozt, 2016-12-15:
It appears to be starting up fine but It cant seem to read data from the iio:device.  This is the new 2.0 build on Kernel 4.9.0

`** (process:2693): DEBUG: Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device2
** (process:2693): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device2 of type ambient light sensor at IIO Buffer Light sensor
** (process:2693): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/trigger3
** (process:2693): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3
** (process:2693): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
** (process:2693): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/trigger3
** (process:2693): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3/scan_elements/in_accel_y_en
** (process:2693): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3/scan_elements/in_accel_x_en
** (process:2693): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device3/scan_elements/in_accel_z_en
** (process:2693): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2693): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2693): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2693): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2693): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2693): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2693): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/trigger2
** (process:2693): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device2/scan_elements/in_intensity_both_en
** (process:2693): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2693): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:2693): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2693): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:2693): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2693): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:2693): DEBUG: No new data available on 'iio:device2'
** (process:2693): DEBUG: No new data available on 'iio:device3'
** (process:2693): DEBUG: No new data available on 'iio:device2'
** (process:2693): DEBUG: No new data available on 'iio:device3'`

sarmbruster, 2016-12-15:
I've found on my X1 Yoga that `monitor-sensor` works nicely on 4.9.0-RC6, but stopped working on 4.9.0-RC8. Using iio-sensor-proxy 1.1-1 from Ubuntu 16.04  here.

rickysarraf, 2016-12-20:
Hmmm. I've got similar issues not creeping into my Yoga 2 13, which was working fine for months. @hellozt  Can you please format your logs ? It is hard to read this way. I'm seeing this on 1.3

The-Compiler, 2016-12-21:
I can't get this to work with the 4.9.0 release (from [here](https://aur.archlinux.org/packages/linux-nvme) with NVMe patches) and iio-sensor-proxy from git either, not even after a suspend/wakeup cycle.

`monitor-sensor` says:

```
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor
```

after suspend:

```
+++ Accelerometer appeared
    Accelerometer orientation changed: undefined
--- Light sensor disappeared
    Light changed: 0.000000 (lux)
```

debug output:

```
** (process:8901): DEBUG: Found input accel at /sys/devices/virtual/input/input15/event9
** (process:8901): DEBUG: Found device /sys/devices/virtual/input/input15/event9 of type accelerometer at Input accelerometer

** (process:8901): WARNING **: Could not open input accel '/dev/input/event9': Operation not permitted
** (process:8901): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:8901): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device

** (process:8901): WARNING **: Could not open input accel '/dev/input/event9': Operation not permitted

** (process:8901): WARNING **: Could not open input accel '/dev/input/event9': Operation not permitted

[...]
```

rannbaron, 2016-12-26:
I've traced the issue to [this kernel patch](https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=d443a0aa3a291e5f78072f2fa464e03bc83fafad). Compiling 4.9-rc8 without it makes screen rotation work again on my X1 Yoga.

rickysarraf, 2016-12-26:
@rannbaron You should mention on the linux-iio mailing list, of this potential regression.

rickysarraf, 2016-12-27:
Reverting that change did not work out for me on 4.9 final with a Lenovo Yoga 2 13.

rannbaron, 2016-12-27:
Huh. Weird. That is the only change between 4.9-rc6 and rc8 that affects iio-sensor-proxy, as far as I can tell.

rannbaron, 2016-12-28:
The chief developer seems to have [a patch to fix this](http://marc.info/?l=linux-iio&m=148285789508236&w=2). I'll test when I have a bit of time.

rannbaron, 2016-12-28:
The chief developer seems to have [a patch to fix this](http://marc.info/?l=linux-iio&m=148285789508236&w=2). I've tested it and it works for me.

rickysarraf, 2016-12-28:
That does not seem to make much different either. @rannbaron

rickysarraf, 2016-12-28:
@rannbaron The change is so generic that, if the correct fix, it should have applied to all devices of the family. Can you please confirm if that patch works on your X1 Yoga ?

rannbaron, 2016-12-28:
@rickysarraf: It might be worth your time to communicate with the developer above. His email is in the link, of course. He was really responsive and helpful when I was first testing his initial patches on LKML.

rickysarraf, 2016-12-28:
Yes. I thought so and just replied to him, and have CCed linux-iio.

hellozt, 2016-12-28:
I can confirm the patch also works for my x1 yoga

hadess, 2017-01-11:
> I can confirm the patch also works for my x1 yoga

Great, closing then.

closed 2017-01-11 by hadess

sarmbruster, 2017-01-13:
Just checked source tree on kernel.org, the referenced patch seems to be merged both for 4.9.3 and 4.10-rc3.

ishitatsuyuki, 2017-01-14:
I'm not sure which sensor are you talking about - nothing works with my 4.9.3.

```
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
```

sarmbruster, 2017-01-14:
On Ubuntu 16.04 with Mainline kernel 4.9.3 it works:
```➜  ~ monitor-sensor 
** Message: Accelerometer orientation changed: right-up
** Message: Light changed: 29.326001 (lux)
** Message: Light changed: 30.624001 (lux)
** Message: Light changed: 34.257002 (lux)
** Message: Accelerometer orientation changed: normal
** Message: Light changed: 7.266000 (lux)
** Message: Light changed: 5.579000 (lux)
** Message: Accelerometer orientation changed: right-up
** Message: Light changed: 35.944002 (lux)
** Message: Light changed: 33.738002 (lux)
** Message: Light changed: 32.181002 (lux)
** Message: Light changed: 31.532001 (lux)
^C
stefan@x1 ➜  ~ uname -a
Linux x1 4.9.3-040903-generic #201701120631 SMP Thu Jan 12 11:33:59 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

sorawee, 2017-01-15:
Somehow installing kernel 4.9.3 doesn't fix the issue for me (got the result just like @ishitatsuyuki). Ubuntu 16.10. I downloaded it from kernel.ubuntu.com/~kernel-ppa/mainline/v4.9.3/

ishitatsuyuki, 2017-01-15:
It seems rather random: after the boot it's unusable, but I found it working after a suspend.

The-Compiler, 2017-01-15:
Also doesn't work for me yet with 4.9.3 on Arch.

Before a suspend I get:

```
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor
```

```
** (process:3482): DEBUG: Found input accel at /sys/devices/virtual/input/input22/event10
** (process:3482): DEBUG: Found device /sys/devices/virtual/input/input22/event10 of type accelerometer at Input accelerometer

** (process:3482): WARNING **: Could not open input accel '/dev/input/event10': Operation not permitted
** (process:3482): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3482): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device

** (process:3482): WARNING **: Could not open input accel '/dev/input/event10': Operation not permitted
```

after the suspend:

```
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor
+++ Accelerometer appeared
    Accelerometer orientation changed: undefined
--- Light sensor disappeared
    Light changed: 0.000000 (lux)
```

fallenpixel, 2017-01-15:
Maintainer of the aur package here: I experience the same but is resolved
in the git version.

The-Compiler, 2017-01-15:
I just reinstalled -git (45f9a63) and still see the same.

sorawee, 2017-01-15:
@ishitatsuyuki you are right! After suspension, I get:

```
$ monitor-sensor
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
    Accelerometer orientation changed: normal
    Light changed: 1.686000 (lux)
    Light changed: 1.038000 (lux)
    Light changed: 1.167000 (lux)
    Accelerometer orientation changed: left-up
```

It works now! So I guess the question is, what's going on with this? Why do we need to suspend it to make it work. Doesn't make much sense...

hadess, 2017-01-16:
> It works now! So I guess the question is, what's going on with this? Why do we need to suspend it to make it work. Doesn't make much sense...

Because the kernel is broken. That's why there's a "driver bug" tag. And the bug is closed because it works for the original reporter.

Depaulicious, 2017-04-16:
So this has been merged in 4.9? I'm on 4.10 and it still works only after suspend.