hadess/iio-sensor-proxy #125: Dell Latitude 7275

sergeycg, 2016-12-22:
v2.0, 4.9 kernel
```
** (process:4343): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/trigger0
** (process:4343): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device0
** (process:4343): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:4343): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/trigger0
** (process:4343): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device0/scan_elements/in_accel_y_en
** (process:4343): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device0/scan_elements/in_accel_x_en
** (process:4343): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device0/scan_elements/in_accel_z_en
** (process:4343): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4343): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4343): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4343): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4343): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4343): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:4343): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:4343): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:4343): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:4343): DEBUG: No new data available on 'iio:device0'
** (process:4343): DEBUG: No new data available on 'iio:device0'
** (process:4343): DEBUG: No new data available on 'iio:device0'
** (process:4343): DEBUG: No new data available on 'iio:device0'
** (process:4343): DEBUG: No new data available on 'iio:device0'
** (process:4343): DEBUG: No new data available on 'iio:device0'
```

**udevadm:**
```
P: /devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device0
N: iio:device0
E: DEVNAME=/dev/iio:device0
E: DEVPATH=/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device0
E: DEVTYPE=iio_device
E: IIO_SENSOR_PROXY_TYPE=iio-buffer-accel
E: MAJOR=244
E: MINOR=0
E: SUBSYSTEM=iio
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=31338267

P: /devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/trigger0
E: DEVPATH=/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/trigger0
E: SUBSYSTEM=iio

P: /devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.17.auto/iio:device1
N: iio:device1
E: DEVNAME=/dev/iio:device1
E: DEVPATH=/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.17.auto/iio:device1
E: DEVTYPE=iio_device
E: IIO_SENSOR_PROXY_TYPE=iio-buffer-accel
E: MAJOR=244
E: MINOR=1
E: SUBSYSTEM=iio
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=31340418

P: /devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.17.auto/trigger1
E: DEVPATH=/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0005/HID-SENSOR-200073.17.auto/trigger1
E: SUBSYSTEM=iio
```

**gdbus:**

```
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = false;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};

```

**monitor-sensor:**

```
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor
```

unfortunately, i can't check it after suspend/wakeup.

Pointedstick, 2016-12-29:
Probably the same as https://github.com/hadess/iio-sensor-proxy/issues/124 and https://github.com/hadess/iio-sensor-proxy/issues/126

sergeycg, 2017-01-14:
accelerometer working after upgrading to 4.9.3 kernel

closed 2017-01-20 by sergeycg