hadess/iio-sensor-proxy #126: HP Spectre x360 Kaby Lake

sauerbraten, 2016-12-25:
Accelerometer orientation is not working on HP Spectre x360 (Kaby Lake, late 2016).

Here's the output of `G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy` (after stopping the service):

```
# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:3062): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.8.auto/trigger0
** (process:3062): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.8.auto/iio:device0
** (process:3062): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.8.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:3062): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.8.auto/trigger0
** (process:3062): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.8.auto/iio:device0/scan_elements/in_accel_y_en
** (process:3062): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.8.auto/iio:device0/scan_elements/in_accel_x_en
** (process:3062): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.8.auto/iio:device0/scan_elements/in_accel_z_en
** (process:3062): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3062): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3062): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3062): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3062): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3062): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:3062): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:3062): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:3062): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3062): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:3062): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3062): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:3062): DEBUG: No new data available on 'iio:device0'
** (process:3062): DEBUG: No new data available on 'iio:device0'
```

No matter what I do, it never finds new data. This is after suspending and resuming, too.

This is the output of `gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy`:
```
$ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = false;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};
```

I'm using Kernel 4.10.0-0.rc0.git9.1.fc26.x86_64 on Fedora 25. Installing 4.10 over the 4.8 kernel in the official repos gave me support for the screen position sensor, so when I fold back my screen, keyboard and touchpad are disabled.

sauerbraten, 2016-12-25:
Overall, this case is very similar to #124 it seems.

Pointedstick, 2016-12-29:
I see the issue as well with the same hardware and kernel. Keyboard and touchpad are correctly disabled when the screen is folded into tent or tablet modes, but autorotation doesn't work.

mgielda, 2017-01-14:
I can confirm that on my Kaby Lake x360 kernel 4.9.3 as well as 4.10rc3 rotation works as per @sarmbruster's comment in #124

sauerbraten, 2017-01-14:
You're right.
In #124 they confirm the Ubuntu kernel working, and I can confirm it works with Fedora's 4.10.0-0.rc3.git1.1.fc26.x86_64. The latest version available in Fedora 25 repositories is 4.8.16, so in case somebody doesn't want to wait, you can get it here: https://apps.fedoraproject.org/packages/kernel

closed 2017-01-14 by sauerbraten

mgielda, 2017-01-15:
There's a weird quirk under wayland BTW, the toushcreen event space does not get rotated so my touch events end up in the wrong place while rotated ;) guess we'll have to wait until Gnome 3.24 unless someone knows a solution...

sauerbraten, 2017-01-15:
Mhmm, just tried it by flipping the screen back into tent mode. Touch events worked fine (Fedora 25, 4.10.0-0.rc3 kernel, gnome-shell 3.22.2, Wayland).