hadess/iio-sensor-proxy #131: cant get iio-sensor-proxy to show anything

vadali, 2016-12-30:
Hi,
I have an hp envy x360, and I am trying to use monitor-sensor to get feedback when the lid is folded, however nothing comes up. I tried compiling from scratch, nothing. 

Doing `udevadm info --export-db` does not give me the right output, I get 
```
P: /devices/platform/coretemp.0/hwmon/hwmon1
E: DEVPATH=/devices/platform/coretemp.0/hwmon/hwmon1
E: SUBSYSTEM=hwmon
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=7589380
```

What could I do? 

I tried googling to see if I can find something that shows me when the lid is folded, but I cant find anything.. Any tips on how else I could know this?

Thanks!

hadess, 2016-12-30:
iio-sensor-proxy doesn't show the lid state, because the lid isn't a sensor.

logind in systemd handles the lid on Linux systems. You can also try using the information from upower (`upower -d | grep -i lid`) though that interface is likely to be removed at some point in the future.

The laptop probably has sensors, but your kernel doesn't have drivers for it. Ask your distribution about them.

closed 2016-12-30 by hadess

vadali, 2016-12-30:
Oh, thanks for explaining. I was under the impression that this tool helps to know the position of the lid (read it [here](https://linuxappfinder.com/blog/auto_screen_rotation_in_ubuntu))

What you suggested about upower does not really help, it only shows if its open/close. I would like to know if the laptop is being put into "tablet mode", so I could disable the keyboard and touchpad and rotate the screen. 

If you have any more information that you think could help me in my quest, please let me know.

Happy new year and thanks!

hadess, 2016-12-30:
The accelerometer would know how the device is turned, not which function it's currently in. For those, you'd need a kernel driver to access the firmware. Disabling the keyboard and touchpad would usually be done by the firmware on most devices.

vadali, 2016-12-30:
Ok, thanks!

Shouldnt my laptop have an accelerometer? (hp envy x360)
What would be your guess as to why I am getting nothing at all?

I tried with kernel 4.0.4 and 4.8.15, but neither gave me anything..

hadess, 2016-12-30:
You probably need the ISH driver, which is new in 4.9. Again, please ask your distribution.

842Mono, 2017-12-22:
@shlomiv any progress with this?