hadess/iio-sensor-proxy #132: A small patch is required for the Thinkpad Yoga 260

jaksi, 2017-01-09:
The accelerometer on the Yoga 260 only works with the following hack:
```patch
diff --git a/src/drv-iio-buffer-accel.c b/src/drv-iio-buffer-accel.c
index ebf94de..16f41ad 100644
--- a/src/drv-iio-buffer-accel.c
+++ b/src/drv-iio-buffer-accel.c
@@ -160,6 +160,7 @@ read_orientation (gpointer user_data)
 static gboolean
 iio_buffer_accel_discover (GUdevDevice *device)
 {
+       return FALSE;
        char *trigger_name;
 
        if (g_strcmp0 (g_udev_device_get_property (device, "IIO_SENSOR_PROXY_TYPE"), "iio-buffer-accel") != 0)
```

I'm using the current master (45f9a63885d6a7b2d881a327afe46ff266c6e134).

hadess, 2017-01-10:
This is obviously wrong. What's the error you get without this patch?

jaksi, 2017-01-10:
Logs using master:
```
systemd[1]: Starting IIO Sensor Proxy service...
iio-sensor-prox[402]: Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device3
iio-sensor-prox[402]: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
iio-sensor-prox[402]: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/trigger4
iio-sensor-prox[402]: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device4
iio-sensor-prox[402]: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device4 of type accelerometer at IIO Buffer accelerometer
iio-sensor-prox[402]: Found IIO buffer compass at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device2
iio-sensor-prox[402]: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device2 of type compass at IIO Buffer Compass
iio-sensor-prox[402]: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/trigger4
iio-sensor-prox[402]: Enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device4/scan_elements/in_accel_y_en
iio-sensor-prox[402]: Enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device4/scan_elements/in_accel_x_en
iio-sensor-prox[402]: Enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device4/scan_elements/in_accel_z_en
iio-sensor-prox[402]: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/trigger3
iio-sensor-prox[402]: Enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device3/scan_elements/in_intensity_both_en
iio-sensor-prox[402]: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/trigger2
iio-sensor-prox[402]: Enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device2/scan_elements/in_magn_z_en
iio-sensor-prox[402]: Enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device2/scan_elements/in_magn_y_en
iio-sensor-prox[402]: Enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device2/scan_elements/in_rot_from_north_magnetic_tilt_comp_en
iio-sensor-prox[402]: Enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device2/scan_elements/in_magn_x_en
iio-sensor-prox[402]: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Built channel array for in_magn_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
iio-sensor-prox[402]: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
[... cut ...]
iio-sensor-prox[402]: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
iio-sensor-prox[402]: Accel read from IIO on 'accel_3d': 23437, -935546, -187500 (scale 0.000010)
iio-sensor-prox[402]: Accel sent by driver (quirk applied): 23437, -935546, -187500 (scale: 0.000010)
iio-sensor-prox[402]: Accel read from IIO on 'accel_3d': 21484, -935546, -193359 (scale 0.000010)
iio-sensor-prox[402]: Accel sent by driver (quirk applied): 21484, -935546, -193359 (scale: 0.000010)
```

hadess, 2017-01-11:
That's most likely a kernel bug, the same one that happens on all the other laptops and tablets as well.

jaksi, 2017-01-16:
Hm, it works after suspending and resuming, so this must be a duplicate of #124

hadess, 2017-01-17:
Very likely indeed. Closing this one.

closed 2017-01-17 by hadess