hadess/iio-sensor-proxy #133: Lenovo Thinkpad Yoga

jwmay2012, 2017-01-10:
`Linux thinkpad-yoga 4.4.39-1-MANJARO #1 SMP PREEMPT Thu Dec 15 21:29:27 UTC 2016 x86_64 GNU/Linux`

```
[jwmay@thinkpad-yoga ~]$ monitor-sensor
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
```

```
[jwmay@thinkpad-yoga ~]$ sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:5306): DEBUG: Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200041.8.auto/iio:device3
** (process:5306): DEBUG: Found device /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200041.8.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
** (process:5306): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.3.auto/trigger0
** (process:5306): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.3.auto/iio:device0
** (process:5306): DEBUG: Found device /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.3.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:5306): DEBUG: Found IIO buffer compass at /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.5.auto/iio:device2
** (process:5306): DEBUG: Found device /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.5.auto/iio:device2 of type compass at IIO Buffer Compass
** (process:5306): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.3.auto/trigger0
** (process:5306): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.3.auto/iio:device0/scan_elements/in_accel_x_en
** (process:5306): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.3.auto/iio:device0/scan_elements/in_accel_y_en
** (process:5306): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.3.auto/iio:device0/scan_elements/in_accel_z_en
** (process:5306): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200041.8.auto/trigger3
** (process:5306): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200041.8.auto/iio:device3/scan_elements/in_intensity_both_en
** (process:5306): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.5.auto/trigger2
** (process:5306): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.5.auto/iio:device2/scan_elements/in_magn_x_en
** (process:5306): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.5.auto/iio:device2/scan_elements/in_magn_y_en
** (process:5306): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.5.auto/iio:device2/scan_elements/in_magn_z_en
** (process:5306): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200083.5.auto/iio:device2/scan_elements/in_rot_from_north_magnetic_tilt_comp_en
** (process:5306): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Built channel array for in_magn_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: index: 3, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5306): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:5306): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:5306): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:5306): DEBUG: No new data available on 'iio:device3'
** (process:5306): DEBUG: No new data available on 'iio:device0'
** (process:5306): DEBUG: No new data available on 'iio:device3'
** (process:5306): DEBUG: No new data available on 'iio:device0'
** (process:5306): DEBUG: No new data available on 'iio:device3'
** (process:5306): DEBUG: No new data available on 'iio:device0'
```

```
P: /devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.3.auto/iio:device0
N: iio:device0
E: DEVNAME=/dev/iio:device0
E: DEVPATH=/devices/pci0000:00/0000:00:1d.0/usb1/1-1/1-1.7/1-1.7:1.0/0003:0483:91D1.0001/HID-SENSOR-200073.3.auto/iio:device0
E: DEVTYPE=iio_device
E: IIO_SENSOR_PROXY_TYPE=iio-buffer-accel
E: MAJOR=246
E: MINOR=0
E: SUBSYSTEM=iio
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=2884494
[... 80 lines cut ...]
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};
```

hadess, 2017-01-10:
What am I supposed to do with this?

jwmay2012, 2017-01-26:
I was just following this from the readme.

> If that doesn't work, please file an issue, make sure any running iio-sensor-proxy has been stopped: systemctl stop iio-sensor-proxy.service and attach the output of: G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy running as root.

I had restarted at least 2 times after installing and when I was testing. But I started it up again at a later time and it was just working. No updates installed or anything and it's been working properly ever since even after multiple reboots. 

Not sure what I was doing wrong, but it seems I ultimately made this issue in error.

closed 2017-01-26 by jwmay2012