hadess/iio-sensor-proxy #134: Accelerometer in Carbon X1 4th - Could not open input accel

pszypowicz, 2017-01-11:
Hi, on Fedora 25 I get this error:

```# G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:19317): DEBUG: Found input accel at /sys/devices/virtual/input/input15/event14
** (process:19317): DEBUG: Found device /sys/devices/virtual/input/input15/event14 of type accelerometer at Input accelerometer

** (process:19317): WARNING **: Could not open input accel '/dev/input/event14': Operation not permitted
** (process:19317): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:19317): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:19317): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:19317): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device

** (process:19317): WARNING **: Could not open input accel '/dev/input/event14': Operation not permitted

** (process:19317): WARNING **: Could not open input accel '/dev/input/event14': Operation not permitted
```
```
# ls -la /dev/input/event14
crw-rw----. 1 root input 13, 78 Jan  9 21:37 /dev/input/event14
```
```
# monitor-sensor 
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor
```

Selinux is not the issue here, as error occurs even with permissive mode. I would appreciate any help with debugging current state :)

hadess, 2017-01-21:
What's the driver used by that accelerometer? This should be mentioned in the output of `udevadm info --export-db`. Looks for "ACCEL" in the output and paste it here.

karolszk, 2017-01-22:
Hi! On debian-testing on the Lenovo Carbon 4 I am sending more details:

From 
udevadm info --export-db

karol@debian:~$ udevadm info --export-db|grep ACCEL
E: POINTINGSTICK_CONST_ACCEL=1.0
E: ID_INPUT_ACCELEROMETER=1
E: ID_INPUT_ACCELEROMETER=1
E: ID_INPUT_ACCELEROMETER=1

karol@debian:~$ sudo dmesg|grep accel
[   11.147502] input: Acer BMA150 accelerometer as /devices/virtual/input/input18

hadess, 2017-01-22:
This is a bug in the kernel. Which might be fixed by:
http://www.spinics.net/lists/platform-driver-x86/msg09702.html

This is the Fedora bug for it:
https://bugzilla.redhat.com/show_bug.cgi?id=1401614

@karolszk you'll want to ask your distribution whether they ship their kernel with that patch, and have them add it if not.

closed 2017-01-22 by hadess

pszypowicz, 2017-01-23:
Hi,

I just wanted to confirm that indeed it was a kernel bug and -  at least in fedora - it is already merged and fixed. 

Thanks @hadess :)

rickysarraf, 2017-01-25:
http://bugs.debian.org/cgi-bin/bugreport.cgi?bug=852596

The same has been reported twice on Debian. I hope the fix gets backported to 4.9 tree, to get into most distributions.

@hadess Regarding the report about system logs being flooded with iio-sensor-proxy messages, shouldn't the logging be rate limited during normal operation ?

hadess, 2017-01-25:
No, it's a plain kernel bug, there's nothing that we can do without adding much more code. Just not worth it.

karolszk, 2017-01-27:
Hm, what will stop working on my thinkpad carbon 4th when I disable the iio-sensor-proxy? Debian maintainter of iio-s-p set won'tfix tag..

hadess, 2017-01-27:
It shouldn't be won't fix, it should be reassigned to the kernel package, it's a kernel bug.

wagenadl, 2017-10-03:
Affects me as well on Ubuntu 17.04.

hadess, 2017-10-04:
It's a kernel bug, so no need to add "me too"'s to this list, thanks.

hadess, 2017-10-04:
@matthiasclasen can you reopen the Fedora kernel bug and follow-up please?