hadess/iio-sensor-proxy #140: ii-sensor-proxy on lenovo carbon 4 fills /var/log/syslog

karolszk, 2017-01-21:
Hello,

I'm using debian-testing with iio-sensor-proxy

iio-sensor-proxy (2.0-1) unstable; urgency=medium

  * [e44c466] New upstream version 2.0
  * [866a7e6] Add libgtk-3-dev to build dependency

 -- Ritesh Raj Sarraf <rrs@debian.org>  Wed, 21 Dec 2016 22:32:51 +0530

and noticed very frequently messages appears on my /var/log/syslog:

Jan 21 23:02:49 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:50 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:51 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:52 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:52 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:53 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:54 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:54 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:55 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:56 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:56 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:57 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:58 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:59 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:02:59 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:03:00 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:03:01 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:03:01 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:03:02 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:03:03 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:03:03 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:03:04 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:03:05 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:03:06 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted
Jan 21 23:03:06 debian iio-sensor-prox[719]: Could not open input accel '/dev/input/event17': Operation not permitted

why?

Regards, Karol

hadess, 2017-01-21:
Is something like AppArmor enabled on your system? iio-sensor-proxy runs as root, so I don't see why it wouldn't be allowed to open that file. Might also want to ask your distribution.

hadess, 2017-01-21:
Seems to also happen on Fedora, closing as a duplicate of #134.

closed 2017-01-21 by hadess

karolszk, 2017-01-21:
Hi @hadess 

karol@debian:~$ ls -la /dev/input/event17 
crw-rw---- 1 root input 13, 81 sty 21 23:20 /dev/input/event17

and indeed iio-sensor-proxy runs as root debian.

Regards, Karol

hadess, 2017-01-22:
Please continue the discussion in the other bug, and provide the information requested there if you can.