hadess/iio-sensor-proxy #141: Inverted laptop screen

Forage, 2017-01-26:
Hi,

After [one bug](https://bugs.launchpad.net/ubuntu/+source/iio-sensor-proxy/+bug/1637864) was fixed for Ubuntu, which made iio-sensor-proxy actually start working, a [different bug](https://bugs.launchpad.net/ubuntu/+source/iio-sensor-proxy/+bug/1644786) appeared that apparently only applies to certain systems. Since I got no reply so far and because I expect it to be a more low level issue to begin with I wanted to try my luck here.

In your documentation you mention `When the accelerometer is not mounted the same way as the screen, we need to modify the readings from the accelerometer to make sure that the computed orientation matches the screen one`. I have the feeling that this might be the case with my laptop. It is, however, unclear to me how to fix it in this case.

My laptop screen is inverted, with the screen upside down and the mouse pointer movement mirrored. 
A rotation lock button is also added to the bottom of the main menu (GNOME Shell 3.22) at the top left, between the settings and session lock button. My laptop has no option to rotate the screen though.

It concerns a HP Envy 15-as006nf.

Is it the correction matrix I need to correct this and if yes, how do I do it exactly?

hadess, 2017-01-26:
> Is it the correction matrix I need to correct this and if yes, how do I do it exactly?

It is. Read the documentation that's linked in the README for details on how to do it.

closed 2017-01-26 by hadess

Forage, 2017-01-26:
I did read it before I posted my questions and it does not provide me with sufficient information to do it.

hadess, 2017-01-26:
Did you read through https://github.com/systemd/systemd/blob/master/hwdb/60-sensor.hwdb ?

Forage, 2017-01-26:
Yes, exactly. I have no idea how to determine the `parent device modalias`, `dmi string` or the matrix itself.

hadess, 2017-01-26:
You'll need to bring that up on a systemd list, this isn't an iio-sensor-proxy specific question, and I'm afraid that I don't have time to field them.