hadess/iio-sensor-proxy #142: X1 Yoga not detecting orientation or light sensor values

alexwaibel, 2017-01-27:
I am running Arch and kernel 4.9 just got pushed to core repos so I updated in hopes that the new patches mentioned in previous issue reports would fix it but now when I run `monitor-sensor` I get this output:
```
Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
```

Additionally, here's the output of 'sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy'

```
** (process:1685): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device3
** (process:1685): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/iio:device3 of type ambient light sensor at IIO Buffer Light sensor
** (process:1685): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device4
** (process:1685): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/iio:device4 of type accelerometer at IIO Buffer accelerometer
** (process:1685): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device2
** (process:1685): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/iio:device2 of type compass at IIO Buffer Compass
** (process:1685): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.9.auto/trigger4
** (process:1685): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200041.2.auto/trigger3
** (process:1685): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.13.auto/trigger2
** (process:1685): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Built channel array for in_magn_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: index: 3, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1685): DEBUG: iio-sensor-proxy is already running, or it cannot own its D-Bus name. Verify installation.
```

grensjo, 2017-01-28:
I am also running Arch with Linux 4.9.6 on my Lenovo Thinkpad X1 Yoga and can confirm this problem: accelerometer is not working, and light sensor starts working only after suspend and resume.

My output from `monitor-sensor` is exactly the same as @alexwaibel's, but the output from `sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy` is slightly different:

```
** (process:1626): DEBUG: Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200041.2.auto/iio:device2
** (process:1626): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200041.2.auto/iio:device2 of type ambient light sensor at IIO Buffer Light sensor
** (process:1626): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/trigger3
** (process:1626): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3
** (process:1626): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
** (process:1626): DEBUG: Found IIO buffer compass at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/iio:device4
** (process:1626): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/iio:device4 of type compass at IIO Buffer Compass
** (process:1626): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/trigger3
** (process:1626): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3/scan_elements/in_accel_y_en
** (process:1626): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3/scan_elements/in_accel_x_en
** (process:1626): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3/scan_elements/in_accel_z_en
** (process:1626): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200041.2.auto/trigger2
** (process:1626): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200041.2.auto/iio:device2/scan_elements/in_intensity_both_en
** (process:1626): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/trigger4
** (process:1626): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/iio:device4/scan_elements/in_magn_z_en
** (process:1626): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/iio:device4/scan_elements/in_magn_y_en
** (process:1626): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/iio:device4/scan_elements/in_rot_from_north_magnetic_tilt_comp_en
** (process:1626): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/iio:device4/scan_elements/in_magn_x_en
** (process:1626): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
[... cut ...]
** (process:1626): DEBUG: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1626): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: index: 3, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
```

See also previous discussion in #71. Please let me know if there is anything I can do to help debugging this issue.

eages, 2017-01-29:
I was getting the same symptoms in Manjaro with kernel 4.9.2 on my Yoga S1. Just trying out kernel 4.10rc3-1 and everything seems to be working without needing to suspend.

alexwaibel, 2017-02-06:
I just upgraded to kernel version 4.9.7 and the issue persists.

peterhoeg, 2017-02-09:
I am seeing the same problem on a yoga 2 pro. Suspending and resuming makes accelerometer and  light sensor work.

turbochris510, 2017-02-11:
Another with the same problem - only works after suspend then resume. Yoga X1 Arch 4.9.8-1

grensjo, 2017-03-11:
This works as expected now on my Lenovo Thinkpad X1 Yoga, running Arch with Linux 4.10.1!

peterhoeg, 2017-03-11:
And my yoga 2 pro is also fine on at least 4.10.1 (I didn't try 4.10).

alexwaibel, 2017-03-13:
I just upgraded to 4.10.1 on Arch and it still does not work for me.

eages, 2017-03-13:
Thinkpad Yoga S1 here with kernel 4.10.1. Running into inconsistencies. It's working right now straight after a reboot but yesterday it wouldn't work after several reboots. Perhaps the sleep workaround needs to be longer? I'll keep an eye on it over the coming days.

alexwaibel, 2017-04-06:
I've continued to notice inconsistencies. My laptop will sometimes rotate while in laptop mode and then get stuck rotated. It also does not always rotate.

jorvis, 2017-07-23:
New Lenovo Yoga X1 (version 2) and same issue here.  No rotation at all and monitor_sensor says:

```
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)

```

Ubuntu Gnome 17.04, kernel 4.10.0-28-generic and iio-sensor-proxy:

```
$ apt-show-versions iio-sensor-proxy
iio-sensor-proxy:amd64/zesty 2.2-0ubuntu1 uptodate
iio-sensor-proxy:i386 not installed

```

hadess, 2017-08-16:
From https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#known-problems:

> Every Linux kernel from 4.3 up to version 4.12 had a bug that made
> made iio-sensor-proxy fail to see any events coming from sensors until the 
> sensor was power-cycled (unplugged and replugged, or suspended and resumed).
> 
> The bug was finally fixed in [this commit](https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/commit/?h=fixes-togreg&id=eafad73ed3851707fa6e3124a255fc049ff9545d)
> in the upstream kernel and backported to stable releases. If you experience
> unresponsive sensors, ask your distributor to make sure this patch was 
> applied to the version you're using.

Please reopen if you still have problems with a kernel containing this fix.

closed 2017-08-16 by hadess

alexwaibel, 2017-08-16:
I'm running kernel 4.12.6 so I believe I should have the patch. I continue to see the "No accelerometer" message:

`monitor-sensor
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== No accelerometer
=== Has ambient light sensor (value: 244.087012, unit: lux)
    Light changed: 244.346012 (lux)
    Light changed: 243.049012 (lux)
    Light changed: 243.308012 (lux)
`

This is the case even after suspend and resume.