hadess/iio-sensor-proxy #145: Doesn't work on Lenovo Yoga 510

ammen99, 2017-02-05:
I noticed that the rotation on my laptop doesn't work. After reading in the internet, I found this repo and tried it. However, it doesn't work. As you have requested, I'm pasting the output of /usr/sbin/iio-sensor-proxy
http://pastebin.com/ZhTxvWfs
This is the output when I start it and then rotate my screen in 360 degrees, so no matter what, there must have been some output. I will be happy to assist with any information you may require.

ammen99, 2017-02-06:
As you labeled it driver bug does that mean I have to file a bug report somewhere in kernel? If yes, which driver?

hadess, 2017-02-06:
It's a bug in the kernel since kernel 4.2. It's being worked on.

hadess, 2017-08-16:
From https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#known-problems:

> Every Linux kernel from 4.3 up to version 4.12 had a bug that made
> made iio-sensor-proxy fail to see any events coming from sensors until the 
> sensor was power-cycled (unplugged and replugged, or suspended and resumed).
> 
> The bug was finally fixed in [this commit](https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/commit/?h=fixes-togreg&id=eafad73ed3851707fa6e3124a255fc049ff9545d)
> in the upstream kernel and backported to stable releases. If you experience
> unresponsive sensors, ask your distributor to make sure this patch was 
> applied to the version you're using.

Please reopen if you still have problems with a kernel containing this fix.

closed 2017-08-16 by hadess