hadess/iio-sensor-proxy #146: problem compiling

chrplr, 2017-02-06:
Hello, 

I must make a silly mistake but I am not able to compile the package.

First I ran aclocal, automake and autoconf to generate a configure script.

But then, when I run it:

~/Downloads/iio-sensor-proxy$ ./configure --prefix=/usr --sysconfdir=/etc
./configure: line 1904: syntax error near unexpected token `git-directory'
./configure: line 1904: `AX_IS_RELEASE(git-directory)'

hadess, 2017-02-06:
You'll need to install `autoconf-archive`. Any recent version of it should have the necessary macros.

closed 2017-02-06 by hadess

chrplr, 2017-02-08:
Thanks! Now, with all the correct dependencies, I could launch 

$ bash ./autogen.sh --prefix=/usr --sysconfdir=/etc

However, make crashes :

make[2]: Entering directory '/home/cp983411/Downloads/iio-sensor-proxy/src'
  CC       monitor_sensor-monitor-sensor.o
  CCLD     monitor-sensor
  CC       fake_input_accelerometer-fake-input-accelerometer.o
fake-input-accelerometer.c: In function ‘send_uinput_event’:
fake-input-accelerometer.c:90:2: error: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Werror=unused-result]
  write (data->uinput, &ev, sizeof(ev));
  ^
fake-input-accelerometer.c:94:2: error: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Werror=unused-result]
  write (data->uinput, &ev, sizeof(ev));
  ^
fake-input-accelerometer.c:98:2: error: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Werror=unused-result]
  write (data->uinput, &ev, sizeof(ev));
  ^
fake-input-accelerometer.c:104:2: error: ignoring return value of ‘write’, declared with attribute warn_unused_result [-Werror=unused-result]
  write (data->uinput, &ev, sizeof(ev));
  
I modified the source code at several places to assign the results of write and fscanf to dummy variables, and could compile the whole thing.

However, there must be a better way to prevent  "unused-results" to generate an error. 
I tried to add -Wno-error=unused-result to  WARN_CFLAG in the Makefile but the compilation still crashed.  Any idea? Thanks!