hadess/iio-sensor-proxy #147: Assertion on startup

hadess, 2017-02-09:
From https://github.com/systemd/systemd/issues/5160
```
Feb  3 09:13:34 ABO-HP iio-sensor-proxy[1101]: ERROR:iio-sensor-proxy.c:185:send_dbus_event: assertion failed: (data->connection)
Feb  3 09:13:34 ABO-HP systemd[1]: iio-sensor-proxy.service: Main process exited, code=killed, status=6/ABRT
Feb  3 09:13:34 ABO-HP systemd[1]: Failed to start IIO Sensor Proxy service.
Feb  3 09:13:34 ABO-HP systemd[1]: iio-sensor-proxy.service: Unit entered failed state.
Feb  3 09:13:34 ABO-HP systemd[1]: iio-sensor-proxy.service: Failed with result 'signal'.
```

Forage, 2017-02-09:
Is there any additional information I can provide to track it down?

hadess, 2017-02-09:
A backtrace of the assertion would be useful, but it's more likely that I just need to stare at the code.

hadess, 2017-02-15:
It's a simple race, if the D-Bus setup is slower than setting up the devices, or we can't own the name quickly enough for whatever reason, then we'd assert when sending out the first changes through D-Bus.

The fix would be to postpone the D-Bus changes until we own the D-Bus name. I have a local patch that needs testing.

Forage, 2017-02-15:
I can test it. You wouldn't happen to have (a PPA with) a build, would you? ;-)

closed 2017-02-15 by hadess

Forage, 2017-02-21:
I can confirm that installing the 2.2 fixes the problem for me. There are no longer instability issues when starting Ubuntu. Thank you!

commit 2ef06018 (2017-02-15): main: Fix possible crash on start due to race

We used to look for sensor devices at the same time as we were setting
up D-Bus. If D-Bus was ready when we finished probing for devices, then
we'd send a D-Bus event. But we would also try to send a D-Bus event
when we got our first event from the sensors, at which point D-Bus might
not be ready.

We really shouldn't be trying to probe and setup the sensors before we
know we can own the D-Bus name, as that might mean that another service
is already running, and we might just break the existing daemon.

Closes #147