hadess/iio-sensor-proxy #150: No new sensor data available on Asus Vivobook Flip TP301

zenitraM, 2017-02-28:
```
$ sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:28255): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/trigger0
** (process:28255): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/iio:device0
** (process:28255): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:28255): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/trigger0
** (process:28255): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/iio:device0/scan_elements/in_accel_y_en
** (process:28255): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/iio:device0/scan_elements/in_accel_x_en
** (process:28255): DEBUG: Already enabled sensor 
/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/iio:device0/scan_elements/in_accel_z_en
** (process:28255): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:28255): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:28255): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:28255): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:28255): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:28255): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:28255): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:28255): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:28255): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
** (process:28255): DEBUG: No new data available on 'iio:device0'
```

while monitor-sensor outputs:
```
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor
```

`udevadm info --export-db` shows:
```
P: /devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/iio:device0
N: iio:device0
E: DEVNAME=/dev/iio:device0
E: DEVPATH=/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/iio:device0
E: DEVTYPE=iio_device
E: IIO_SENSOR_PROXY_TYPE=iio-buffer-accel
E: MAJOR=243
E: MINOR=0
E: SUBSYSTEM=iio
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=18638331

P: /devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/trigger0
E: DEVPATH=/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/trigger0
E: SUBSYSTEM=iio
```

Sensor is detected but with `undefined` orientation:
```
$ gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy
node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = false;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};
```

Seems similar to #117, however reading the raw values from `/sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.5.auto/iio:device0/*raw` works and returns real changing values. Suspending and resuming doesn't change anything.

Any way I can help to further debug this?

zenitraM, 2017-02-28:
Hackily-disabling the trigger device for the accelerometer seems to make it work:

```diff
diff --git a/src/drv-iio-buffer-accel.c b/src/drv-iio-buffer-accel.c
index ebf94de..d93778c 100644
--- a/src/drv-iio-buffer-accel.c
+++ b/src/drv-iio-buffer-accel.c
@@ -113,6 +113,7 @@ bail:
 static char *
 get_trigger_name (GUdevDevice *device)
 {
+       return NULL;
        GList *devices, *l;
        GUdevClient *client;
        gboolean has_trigger = FALSE;
```

Depaulicious, 2017-04-14:
The same issue happens on a Lenovo Thinkpad X1 Yoga. While it worked for a while, at some point I started getting this issue.
In my case the ambient light sensor also stopped working, I always get 0.0 lux. My logs:

```
$ sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:13704): DEBUG: Found IIO buffer ALS at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200041.2.auto/iio:device4
** (process:13704): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200041.2.auto/iio:device4 of type ambient light sensor at IIO Buffer Light sensor
** (process:13704): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/trigger3
** (process:13704): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3
** (process:13704): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3 of type accelerometer at IIO Buffer accelerometer
** (process:13704): DEBUG: Found IIO buffer compass at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/iio:device2
** (process:13704): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/iio:device2 of type compass at IIO Buffer Compass
** (process:13704): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/trigger3
** (process:13704): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3/scan_elements/in_accel_y_en
** (process:13704): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3/scan_elements/in_accel_x_en
** (process:13704): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3/scan_elements/in_timestamp_en
** (process:13704): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3/scan_elements/in_accel_z_en
** (process:13704): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:13704): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0

** (process:13704): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3/in_scale

** (process:13704): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200073.9.auto/iio:device3/in_offset
** (process:13704): DEBUG: Got type for in_timestamp: is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:13704): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:13704): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:13704): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:13704): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:13704): DEBUG: Built channel array for in_timestamp: index: 3, is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:13704): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200041.2.auto/trigger4
** (process:13704): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200041.2.auto/iio:device4/scan_elements/in_intensity_both_en
** (process:13704): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:13704): DEBUG: Built channel array for in_intensity_both: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:13704): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/trigger2
** (process:13704): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/iio:device2/scan_elements/in_magn_z_en
** (process:13704): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200083.13.auto/iio:device2/scan_elements/in_magn_y_en
[... cut ...]
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = true;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};
```

Depaulicious, 2017-04-14:
I'd also like to add some info about the `failed to read float from .../io_scale` error. I listed the files in the specified directory and there's no `io_scale`. However there is `io_accel_scale`. I don't know if that's related.

```
$ ls -l   
total 0
drwxr-xr-x 2 root root    0 apr 14 12:35 buffer
-rw-r--r-- 1 root root 4096 apr 14 12:56 current_timestamp_clock
-r--r--r-- 1 root root 4096 apr 14 12:56 dev
-rw-r--r-- 1 root root 4096 apr 14 12:56 in_accel_hysteresis
-rw-r--r-- 1 root root 4096 apr 14 12:35 in_accel_offset
-rw-r--r-- 1 root root 4096 apr 14 12:56 in_accel_sampling_frequency
-rw-r--r-- 1 root root 4096 apr 14 12:35 in_accel_scale
-rw-r--r-- 1 root root 4096 apr 14 12:34 in_accel_x_raw
-rw-r--r-- 1 root root 4096 apr 14 12:34 in_accel_y_raw
-rw-r--r-- 1 root root 4096 apr 14 12:34 in_accel_z_raw
-r--r--r-- 1 root root 4096 apr 14 12:35 name
drwxr-xr-x 2 root root    0 apr 14 12:56 power
drwxr-xr-x 2 root root    0 apr 14 12:34 scan_elements
lrwxrwxrwx 1 root root    0 apr 14 12:35 subsystem -> ../../../../../../../bus/iio
drwxr-xr-x 2 root root    0 apr 14 12:35 trigger
-rw-r--r-- 1 root root 4096 apr 14 12:34 uevent
```

hadess, 2017-08-16:
From https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#known-problems:

> Every Linux kernel from 4.3 up to version 4.12 had a bug that made
> made iio-sensor-proxy fail to see any events coming from sensors until the 
> sensor was power-cycled (unplugged and replugged, or suspended and resumed).
> 
> The bug was finally fixed in [this commit](https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/commit/?h=fixes-togreg&id=eafad73ed3851707fa6e3124a255fc049ff9545d)
> in the upstream kernel and backported to stable releases. If you experience
> unresponsive sensors, ask your distributor to make sure this patch was 
> applied to the version you're using.

Please reopen if you still have problems with a kernel containing this fix.

closed 2017-08-16 by hadess