hadess/iio-sensor-proxy #151: Where to get gio.h and gudev.h?

miltmobley, 2017-03-02:
I am browsing your code for iio-sensor-proxy, and also found your libgudev.

What is the approved way to obtain gio.h and gudev.h? Are they in some gtk/gnome development
package, or must I copy them from libgudev?

BTW, I did clone ZIP on iio*, and found configure file is missing.

Finally, kernel 4.10 now has driver support for the new Intel ISH bus, which provides access to sensors on the processor package (e.g. orientation) via the HID subsystem. Lack of this support appears
to be a big reason why autorotate doesn't work on newer laptops. Are you working on adding this
support to iio-sensor-proxy?

hadess, 2017-03-02:
> I am browsing your code for iio-sensor-proxy, and also found your libgudev.
> 
> What is the approved way to obtain gio.h and gudev.h? Are they in some gtk/gnome development
> package, or must I copy them from libgudev?

Your distribution's gio and gudev development packages.

> BTW, I did clone ZIP on iio*, and found configure file is missing.

Use the releases, or run `./autogen.sh`

> Finally, kernel 4.10 now has driver support for the new Intel ISH bus, which provides access to sensors > on the processor package (e.g. orientation) via the HID subsystem. Lack of this support [...]

They're already supported.

closed 2017-03-02 by hadess

miltmobley, 2017-03-03:
Please explain how they are supported.

I don't see any references to hid in the git code except for two comments
about using microsoft defined hid units. Is the git repo out of date?

The Intel engineer who published https://lwn.net/Articles/690997/
concerning the ish patches said they had a fix involving modifying a
systemd config file.
So far I didn't fine 'iio-sensor-proxy.service' in any of my systemd files
or udev files.

My openSuse system on Dell inspiron 15 is running iio-sensor-proxy v
1.1-1.1. Is this new enough to have the support?