hadess/iio-sensor-proxy #155: ASUS UX330UA support on th way.

olerem, 2017-03-29:
Currently acpi_als can't read actual data on this laptop, acpi is providing only dummy values.
One ACPI bug is preventing it, i started to work on this issue.

olerem, 2017-03-29:
Probably it should be documented here. I have seen it on at least 3 different ASUS models. ux32a, ux302, ux330ua - the DSDT code is almost identical: it provides a switch to enable real values or provdie some static dummy value.

olerem, 2017-05-02:
patch is accepted for testing https://patchwork.kernel.org/patch/9704803/

hadess, 2017-06-12:
Is this to report more laptops to be added to the tested list? We only add devices that were actually tested, not potentially supported devices.

olerem, 2017-06-12:
It is tested, i own it and wrote the quirk :)

commit 02dfc189 (2017-06-12): README: Add Asus UX330UA to the tested devices

Closes: #155
closed 2017-06-12 by hadess