hadess/iio-sensor-proxy #156: Screenrotation on HP Pavilion x2 12 not working

doniks, 2017-03-30:
I have this laptop with a detachable touch screen. I have Ubuntu 16.10 beta release installed and I try to get the display to automatically rotate when I rotate the screen. I think iio-sensor-proxy is the culprit why it is not working.

I paste below/attach the commands I tried so far trying to analyse the problem. Please let me whether I should try anything else. 

```
sudo systemctl status iio-sensor-proxy
● iio-sensor-proxy.service - IIO Sensor Proxy service
   Loaded: loaded (/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: enabled)
   Active: inactive (dead)

mrt 30 19:41:21 hp systemd[1]: Starting IIO Sensor Proxy service...
mrt 30 19:41:24 hp systemd[1]: Started IIO Sensor Proxy service.
mrt 30 20:51:15 hp systemd[1]: Starting IIO Sensor Proxy service...
mrt 30 20:51:18 hp systemd[1]: Started IIO Sensor Proxy service.
peter@hp:~$ systemctl stop iio-sensor-proxy
sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:27919): DEBUG: Could not find any supported sensors
udevadm info --export-db | grep  iio:device0
P: /devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200076.4.auto/iio:device0
N: iio:device0
E: DEVNAME=/dev/iio:device0
E: DEVPATH=/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200076.4.auto/iio:device0
udevadm info --export-db | grep -C 10 iio:device > udev.log
grep -H . $( find /sys/bus/iio/devices/*/ -type f | grep -Eve 'power|buffer|uevent' | sort ) > ~/iio.1
grep: /sys/bus/iio/devices/iio:device1/in_incli_hysteresis: Invalid argument
grep: /sys/bus/iio/devices/iio:device4/in_rot_hysteresis: Invalid argument
# rotate the screen 90 degree left
grep -H . $( find /sys/bus/iio/devices/*/ -type f | grep -Eve 'power|buffer|uevent' | sort ) > ~/iio.2
grep: /sys/bus/iio/devices/iio:device1/in_incli_hysteresis: Invalid argument
grep: /sys/bus/iio/devices/iio:device4/in_rot_hysteresis: Invalid argument
```

doniks, 2017-03-30:
[iio.1.txt](https://github.com/hadess/iio-sensor-proxy/files/883640/iio.1.txt)
[iio.2.txt](https://github.com/hadess/iio-sensor-proxy/files/883641/iio.2.txt)
[udev.log.txt](https://github.com/hadess/iio-sensor-proxy/files/883639/udev.log.txt)

hadess, 2017-06-12:
Please provide the iio-sensor-proxy debug log as mentioned in the README.

closed 2017-08-16 by hadess