hadess/iio-sensor-proxy #157: Quick question

alothings, 2017-04-01:
How do you start this after installation? Just systemctl start iio-sensor-proxy.service ?

I installed it 2 days ago and it was working. Today I switched to the Windows partition for a few hours, and when I logged back onto Manjaro, the screen does not rotate. 
It says iio-sensor-proxy service is running when I run systemctl, but it does not seem to be working. Any way to troubleshoot, or commands to send you the logs?

Thanks!

Houndie, 2017-04-19:
Not sure if this is your problem or not, but "systemctl start" just starts something the one time--you also want "systemctl enable iio-sensor-proxy.service" to have it autostart as part of the boot process.

hadess, 2017-06-12:
Yes, there's nothing special about the `iio-sensor-proxy.service`, it's the same process as for any systemd service.

closed 2017-06-12 by hadess