hadess/iio-sensor-proxy #158: Dell Inspiron 13 5000 does not rotate screen

Windsurfer01, 2017-04-01:
I downloaded and installed iio-sensor-proxy. I also had to load some dependencies. The sensor was then reported. I have simple keystroke commands to rotate screen. I have gnome and ubuntu 16.04 installed, with kernel 4.8.0.44. 

In /sys/bus/acpi/devices/smo8810:00/subsystem/drivers only smo8800 is referred to.

I am happy to run any tests, provided I am given enough guidance.

bravnsgaard, 2017-04-11:
I too have a new Inspiron 13 5000. I'm running Linux Mint Serena (18.1) with a 4.10 series kernel. I do see rotation, but the rotation is consistently wrong. Running monitor-sensor shows that I have bottom-up when I rotate the screen 90 degrees clockwise, and I get right-up when I have the screen in normal orientation. I first suspected that this was a Cinnamon problem, but the output from monitor-sensor suggests that it is either the accelerometer itself or iio-sensor-proxy. Cinnamon just reacts (correctly) to the signal it receives.

I read somewhere that iio-sensor-proxy has problems with kernels above 4.7?

I too am willing to do some testing if needed.

jrbenito, 2017-04-12:
Samething here, Linux Mint Serena and Inspiron 13 5000 (5368). I can test too.

Windsurfer01, 2017-05-04:
I tried Ubuntu 17.4. The accelerometer was detected and the screen appeared to rotate correctly. I no longer have Ubuntu installed, so can't verify it. I'll try kernel 4.10 and report back.

Windsurfer01, 2017-05-09:
I have installed kernel 4.10.0-20 with no problems yet.
I then used the following script [http://www.androidtablets.net/threads/rotating-screen-touchpad-on-linux.33309/](url) I changed 'ILITEK ILITEK Multi-Touch' to 'Raydium Corporation Raydium Touch System' to suit the Dell Inspiron 13 5000. I got this info from xinput --list.
It works well. However:
1) If the panel is set to autohide it is not possible to access the onscreen keyboard. This is because the panel will not show itself. (It may be that I have missed something and am not swiping properly.)
2) The main keyboard remains active.
The modified script is attached. I will look for solutions to the above problems.

Windsurfer01, 2017-05-10:
The output from xinput --list was:= as below.

mike@mike-Inspiron-13-5378 ~ $ xinput --list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Raydium Corporation Raydium Touch System	id=11	[slave  pointer  (2)]
⎜   ↳ DELL0740:00 06CB:7E7E Touchpad          	id=12	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=16	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Power Button                            	id=8	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=9	[slave  keyboard (3)]
    ↳ Integrated_Webcam_HD                    	id=10	[slave  keyboard (3)]
    ↳ Intel Virtual Button driver             	id=13	[slave  keyboard (3)]
    ↳ Intel HID events                        	id=14	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=15	[slave  keyboard (3)]
    ↳ Dell WMI hotkeys                        	id=17	[slave  keyboard (3)]

I am now trying to get a response from iio-sensor-proxy.

jrbenito, 2017-05-10:
@Windsurfer01 

How did you make virtual keyboard work? (and which one are you using?)

I put a script to rotate screen and touchpad too.

thanks

jrbenito, 2017-05-10:
Ah, in the meantime, if I understood the issue correctly, iio-sensor-proxy is not culprit of wrong report. The newer systemd makes use of a correction matrix for the accel that would correct output to iio-sensor-proxy. However, Mint is two version behind (since it is based on Ubuntu 16,04). Actually Ubuntu 17,04 might have the newer version of systemd and the sensor correction for any device is a matter of config file.

Windsurfer01, 2017-05-10:
The on-screen keyboard appears at the bottom of the rotated screen if System Settings|Panel|Auto-hide panel is set to 'Intelligently hide panel'. I am using Linux Mint 18.1 Serena 64 bit with Cinnamon 3.2.7 and kernel 4.10.0-20.

Windsurfer01, 2017-05-10:
The script I used and a screen shot are attached. The script is a mash-up of various other scripts, including yours. The keyboard is the default one. I do not know of any others. I tried to activate arrow keys but it did not work.
[image]
[rotate.txt](https://github.com/hadess/iio-sensor-proxy/files/990428/rotate.txt)

hadess, 2017-06-12:
This really isn't the place to discuss on-screen keyboards. Reopen a new issue and keep it on subject if you still have problems, after having checked the minimum requirements in the README.

closed 2017-06-12 by hadess