hadess/iio-sensor-proxy #159: Rotation stopped (Dell Inspiron 17 7000)

FFortissimo, 2017-04-13:
I got a new Dell Inspiron 17 7000 (2-in-1) a few months ago and installed Fedora 25.
Screen rotation worked out of the box.

This week I noticed the screen rotation not functioning any more.
Of course this gives troubles when using the notebook as tablet or in tent-mode.

I tried restarting iio-sensor-proxy.service, but that didn't help.

`
[root@xx ~]# systemctl status iio-sensor-proxy.service 
● iio-sensor-proxy.service - IIO Sensor Proxy service
   Loaded: loaded (/usr/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: disabled)
   Active: active (running) since do 2017-04-13 20:54:20 CEST; 5s ago
 Main PID: 4458 (iio-sensor-prox)
    Tasks: 4 (limit: 4915)
   CGroup: /system.slice/iio-sensor-proxy.service
           └─4458 /usr/sbin/iio-sensor-proxy

apr 13 20:54:17 xx.xx systemd[1]: Starting IIO Sensor Proxy service...
apr 13 20:54:20 xx.xx systemd[1]: Started IIO Sensor Proxy service.
`
I tried  ´xrandr' but that didn't work either:

`
[jeroen@xx ~]$ xrandr --listmonitors 
Monitors: 1
 0: +XWAYLAND0 1920/380x1080/210+0+0  XWAYLAND0
[jeroen@xx ~]$ xrandr --output XWAYLAND0 --rotate right
xrandr: Configure crtc 0 failed
[jeroen@xx ~]$ sudo xrandr --output XWAYLAND0 --rotate right
No protocol specified
Can't open display :0
`

What do I check/do to get this working again?

FFortissimo, 2017-04-13:
I found something:
Fedora (4.10.8-200.fc25.x86_64) 25 (Workstation Edition)  -> doesn't work
Fedora (4.10.5-200.fc25.x86_64) 25 (Workstation Edition) -> does work
Fedora (4.9.14-200.fc25.x86_64) 25 (Workstation Edition) -> does work

looking into /etc/grub2-efi.cfg I can't find any differece but the version:
> BEGIN /etc/grub.d/10_linux 
menuentry 'Fedora (4.10.8-200.fc25.x86_64) 25 (Workstation Edition)' --class fedora --class gnu-linux --class gnu --class os --unrestricted $menuentry_id_option 'gnulinux-4.8.6-300.fc25.x86_64-advanced-baa43eaa-189c-46c1-92a5-601fd75753cb' {
        load_video
        set gfxpayload=keep
        insmod gzio
        insmod part_gpt
        insmod ext2
        set root='hd0,gpt2'
        if [ x$feature_platform_search_hint = xy ]; then
          search --no-floppy --fs-uuid --set=root --hint-bios=hd0,gpt2 --hint-efi=hd0,gpt2 --hint-baremetal=ahci0,gpt2  7aa3e707-1762-42f4-a7e4-db7b96736289
        else
          search --no-floppy --fs-uuid --set=root 7aa3e707-1762-42f4-a7e4-db7b96736289
        fi
        linuxefi /vmlinuz-4.10.8-200.fc25.x86_64 root=UUID=baa43eaa-189c-46c1-92a5-601fd75753cb ro rootflags=subvol=root rhgb quiet LANG=nl_NL.UTF-8
        initrdefi /initramfs-4.10.8-200.fc25.x86_64.img
}
menuentry 'Fedora (4.10.5-200.fc25.x86_64) 25 (Workstation Edition)' --class fedora --class gnu-linux --class gnu --class os --unrestricted $menuentry_id_option 'gnulinux-4.8.6-300.fc25.x86_64-advanced-baa43eaa-189c-46c1-92a5-601fd75753cb' {
        load_video
        set gfxpayload=keep
        insmod gzio
        insmod part_gpt
        insmod ext2
        set root='hd0,gpt2'
        if [ x$feature_platform_search_hint = xy ]; then
          search --no-floppy --fs-uuid --set=root --hint-bios=hd0,gpt2 --hint-efi=hd0,gpt2 --hint-baremetal=ahci0,gpt2  7aa3e707-1762-42f4-a7e4-db7b96736289
        else
          search --no-floppy --fs-uuid --set=root 7aa3e707-1762-42f4-a7e4-db7b96736289
        fi
        linuxefi /vmlinuz-4.10.5-200.fc25.x86_64 root=UUID=baa43eaa-189c-46c1-92a5-601fd75753cb ro rootflags=subvol=root rhgb quiet LANG=nl_NL.UTF-8
        initrdefi /initramfs-4.10.5-200.fc25.x86_64.img
}
menuentry 'Fedora (4.9.14-200.fc25.x86_64) 25 (Workstation Edition)' --class fedora --class gnu-linux --class gnu --class os --unrestricted $menuentry_id_option 'gnulinux-4.8.6-300.fc25.x86_64-advanced-baa43eaa-189c-46c1-92a5-601fd75753cb' {
        load_video
        set gfxpayload=keep
        insmod gzio
        insmod part_gpt
        insmod ext2
        set root='hd0,gpt2'
        if [ x$feature_platform_search_hint = xy ]; then
          search --no-floppy --fs-uuid --set=root --hint-bios=hd0,gpt2 --hint-efi=hd0,gpt2 --hint-baremetal=ahci0,gpt2  7aa3e707-1762-42f4-a7e4-db7b96736289
        else
          search --no-floppy --fs-uuid --set=root 7aa3e707-1762-42f4-a7e4-db7b96736289
        fi
        linuxefi /vmlinuz-4.9.14-200.fc25.x86_64 root=UUID=baa43eaa-189c-46c1-92a5-601fd75753cb ro rootflags=subvol=root rhgb quiet LANG=nl_NL.UTF-8
        initrdefi /initramfs-4.9.14-200.fc25.x86_64.img
}

FFortissimo, 2017-04-16:
Also added at https://bugzilla.redhat.com/show_bug.cgi?id=1442630

FFortissimo, 2017-04-29:
4.10.9-200.fc25.x86_64 still no rotation

ragibkl, 2017-05-05:
Hi @FFortissimo , I'm curious if your problem is similar to mine. Did you try this workaround?
https://github.com/hadess/iio-sensor-proxy/issues/123#issuecomment-265482710

FFortissimo, 2017-05-18:
Didn't work?

watch: failed to parse argument: '0.1' watch: failed to parse argument: '0.1'

FFortissimo, 2017-06-03:
4.11.3-200.fc25.x86_64 rotation is back :D

ragibkl, 2017-06-04:
> 4.11.3-200.fc25.x86_64 rotation is back :D 

I can confirm this as well. just updated mine and this works again.
yeay!

hadess, 2017-06-08:
> 4.11.3-200.fc25.x86_64 rotation is back :D

Glad it works, but you need to see whether the problem happens without the "sleep" in `iio-sensor-proxy.c` otherwise it's just the kernel bug I've mentioned in other issues is still there, and it just so happens that the race is simply obscured by this kernel build.

ammen99, 2017-06-09:
I have another laptop, rotation indeed works for me but only until I suspend my machine. After resuming it doesn't work again, so I don't think that the bug has been resolved(I also use the new 4.11.3 kernel)

hadess, 2017-08-16:
From https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#known-problems:

> Every Linux kernel from 4.3 up to version 4.12 had a bug that made
> made iio-sensor-proxy fail to see any events coming from sensors until the 
> sensor was power-cycled (unplugged and replugged, or suspended and resumed).
> 
> The bug was finally fixed in [this commit](https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/commit/?h=fixes-togreg&id=eafad73ed3851707fa6e3124a255fc049ff9545d)
> in the upstream kernel and backported to stable releases. If you experience
> unresponsive sensors, ask your distributor to make sure this patch was 
> applied to the version you're using.

Please reopen if you still have problems with a kernel containing this fix.

closed 2017-08-16 by hadess