hadess/iio-sensor-proxy #160: === Has accelerometer (orientation: undefined)

bchernicoff, 2017-04-16:
Ubuntu 17.04 on HP ENVY X360 13t-y000 Convertible Laptop

** (process:10784): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.10.auto/trigger1
** (process:10784): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.10.auto/iio:device1
** (process:10784): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.10.auto/iio:device1 of type accelerometer at IIO Buffer accelerometer
** (process:10784): DEBUG: Found IIO buffer compass at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.19.auto/iio:device3
** (process:10784): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.19.auto/iio:device3 of type compass at IIO Buffer Compass
** (process:10784): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.10.auto/trigger1
** (process:10784): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.10.auto/iio:device1/scan_elements/in_accel_y_en
** (process:10784): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.10.auto/iio:device1/scan_elements/in_accel_x_en
** (process:10784): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.10.auto/iio:device1/scan_elements/in_timestamp_en
** (process:10784): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.10.auto/iio:device1/scan_elements/in_accel_z_en
** (process:10784): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0

** (process:10784): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.10.auto/iio:device1/in_scale

** (process:10784): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.10.auto/iio:device1/in_offset
** (process:10784): DEBUG: Got type for in_timestamp: is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:10784): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Built channel array for in_timestamp: index: 3, is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:10784): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.19.auto/trigger3
** (process:10784): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.19.auto/iio:device3/scan_elements/in_magn_z_en
** (process:10784): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.19.auto/iio:device3/scan_elements/in_magn_y_en
** (process:10784): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.19.auto/iio:device3/scan_elements/in_rot_from_north_magnetic_tilt_comp_en
** (process:10784): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0003/HID-SENSOR-200083.19.auto/iio:device3/scan_elements/in_magn_x_en
** (process:10784): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Built channel array for in_magn_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Built channel array for in_magn_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Built channel array for in_magn_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:10784): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: index: 3, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0

yoshz, 2017-06-07:
Same issue here on Asus Zenbook Flip UX360UAK with Ubuntu 17.04

auscompgeek, 2017-06-10:
@bchernicoff @yoshz Does iio-sensor-proxy work after waking from suspend?

hadess, 2017-06-12:
The version of iiio-sensor-proxy used and the output of `monitor-sensor` would also be helpful.

yoshz, 2017-06-13:
Here is my info:

```
20:04 $ sudo systemctl start iio-sensor-proxy.service 

20:05 $ tail /var/log/syslog | grep iio-sensor
Jun 13 20:05:11 zbook iio-sensor-prox[3085]: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.6.auto/iio:device0/in_scale
Jun 13 20:05:11 zbook iio-sensor-prox[3085]: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.6.auto/iio:device0/in_offset

20:07 $ apt-cache show iio-sensor-proxy
Package: iio-sensor-proxy
Priority: optional
Section: utils
Installed-Size: 100
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Ritesh Raj Sarraf <rrs@debian.org>
Architecture: amd64
Version: 2.2-0ubuntu1
Depends: libc6 (>= 2.7), libglib2.0-0 (>= 2.39.90), libgudev-1.0-0 (>= 146), systemd
Filename: pool/main/i/iio-sensor-proxy/iio-sensor-proxy_2.2-0ubuntu1_amd64.deb
Size: 25936
MD5sum: e291414f13e13145683645904db7603d
SHA1: 32cdcfc918b28ab053ed8fe2979e9bd935cdca44
SHA256: 64e3b6572b1289ecb064edd321dfdb7641b83904c78b2821e0b2b6536ee9434e
Description-en: IIO sensors to D-Bus proxy
 Industrial I/O subsystem is intended to provide support for devices
 that in some sense are analog to digital or digital to analog convertors
 .
 Devices that fall into this category are:
  * ADCs
  * Accelerometers
  * Gyros
  * IMUs
  * Capacitance to Digital Converters (CDCs)
  * Pressure Sensors
  * Color, Light and Proximity Sensors
  * Temperature Sensors
  * Magnetometers
  * DACs
  * DDS (Direct Digital Synthesis)
  * PLLs (Phase Locked Loops)
  * Variable/Programmable Gain Amplifiers (VGA, PGA)
 .
 Many ultrabooks ship with some of these devices. iio-sensor-proxy is
 intended to serve as a proxy, providing access to these devices through
 a D-Bus interface
Description-md5: c51156da1a0cdeb9cf68b6b02bcdc255
Homepage: https://github.com/hadess/iio-sensor-proxy
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Origin: Ubuntu
Supported: 9m
Task: ubuntu-desktop, ubuntu-usb, kubuntu-desktop, kubuntu-full, edubuntu-desktop, edubuntu-usb, ubuntustudio-audio, ubuntu-gnome-desktop, ubuntukylin-desktop, ubuntu-budgie-desktop

20:07 $ monitor-sensor 
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor
```

iio-sensor-proxy starts but with errors.

harriseldon, 2017-06-23:
I saw a similar issue and saw some interesting items in the debug log. 
First I saw this warning in the debug log:
** (process:12139): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0001/HID-SENSOR-200073.7.auto/iio:device3/*in_scale*

I decided to check the referenced sys path to see if in_scale was returning a proper value. I noticed this:

cat /sys/devices/pci0000\:00/0000\:00\:13.0/\{33AECD58-B679-4E54-9BD9-A04D34F0C226\}/001F\:8086\:22D8.0001/HID-SENSOR-200073.7.auto/iio\:device3/
buffer/                      *in_accel_scale*               scan_elements/
current_timestamp_clock      in_accel_x_raw               subsystem/
dev                          in_accel_y_raw               trigger/
in_accel_hysteresis          in_accel_z_raw               uevent
*in_accel_offset*              name                         
in_accel_sampling_frequency  power/        

Notice that in_scale is not present. Could this be an issue that the sys elements changed at a certain kernel level?

I did a cat on in_accel_scale and got 0.000009806

Here is my kernel version:
Linux 4.11.6-1-ARCH #1 SMP PREEMPT Sat Jun 17 08:19:42 CEST 2017 x86_64 GNU/Linux

I am running iio-sensor-proxy 2.2

parkerlreed, 2017-07-26:
Same here on 4.12.3.

````
Jul 26 16:09:20 yoga710 iio-sensor-prox[382]: Failed to read float from /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0001/HID-SENSOR-200073.2.auto/iio:device0/in_scale
Jul 26 16:09:20 yoga710 iio-sensor-prox[382]: Failed to read float from /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-ITE8350:00/0018:048D:8350.0001/HID-SENSOR-200073.2.auto/iio:device0/in_offset
````

parkerlreed, 2017-07-26:
What's weirder is that "in_scale" doesn't even exist in the source

````
[parker@yoga710 iio-sensor-proxy]$ grep -r "in_accel" .
./src/drv-iio-poll-accel.c:	accel_x = sysfs_get_int (data->dev, "in_accel_x_raw");
./src/drv-iio-poll-accel.c:	accel_y = sysfs_get_int (data->dev, "in_accel_y_raw");
./src/drv-iio-poll-accel.c:	accel_z = sysfs_get_int (data->dev, "in_accel_z_raw");
./src/drv-iio-poll-accel.c:	drv_data->scale = g_udev_device_get_sysfs_attr_as_double (device, "in_accel_scale");
./src/drv-iio-buffer-accel.c:	process_scan_1(data.data + or_data->buffer_data->scan_size*i, or_data->buffer_data, "in_accel_x", &accel_x, &scale, &present_x);
./src/drv-iio-buffer-accel.c:	process_scan_1(data.data + or_data->buffer_data->scan_size*i, or_data->buffer_data, "in_accel_y", &accel_y, &scale, &present_y);
./src/drv-iio-buffer-accel.c:	process_scan_1(data.data + or_data->buffer_data->scan_size*i, or_data->buffer_data, "in_accel_z", &accel_z, &scale, &present_z);
./data/80-iio-sensor-proxy.rules:SUBSYSTEM=="iio", TEST=="in_accel_x_raw", TEST=="in_accel_y_raw", TEST=="in_accel_z_raw", ENV{IIO_SENSOR_PROXY_TYPE}="iio-poll-accel"
./data/80-iio-sensor-proxy.rules:SUBSYSTEM=="iio", TEST=="scan_elements/in_accel_x_en", TEST=="scan_elements/in_accel_y_en", TEST=="scan_elements/in_accel_z_en", ENV{IIO_SENSOR_PROXY_TYPE}="iio-buffer-accel"
[parker@yoga710 iio-sensor-proxy]$ grep -r "in_scale" .
[parker@yoga710 iio-sensor-proxy]$ 
````

hadess, 2017-08-16:
From https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#known-problems:

> Every Linux kernel from 4.3 up to version 4.12 had a bug that made
> made iio-sensor-proxy fail to see any events coming from sensors until the 
> sensor was power-cycled (unplugged and replugged, or suspended and resumed).
> 
> The bug was finally fixed in [this commit](https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/commit/?h=fixes-togreg&id=eafad73ed3851707fa6e3124a255fc049ff9545d)
> in the upstream kernel and backported to stable releases. If you experience
> unresponsive sensors, ask your distributor to make sure this patch was 
> applied to the version you're using.

Please reopen if you still have problems with a kernel containing this fix.

closed 2017-08-16 by hadess

parkerlreed, 2017-08-16:
Am I missing something? Pure upstream doesn't seem to have that fix. https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/iio/common/hid-sensors/hid-sensor-trigger.c

There is no `pr_debug` anywhere in there.

parkerlreed, 2017-08-16:
````
[parker@yoga710 ~]$ monitor-sensor 
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== Has ambient light sensor (value: 0.000000, unit: lux)
^C
[parker@yoga710 ~]$ pacman -Q linux-git iio-sensor-proxy-git 
linux-git 4.13rc5.r67.g510c8a899caf-1
iio-sensor-proxy-git r260.d0ced55-1
[parker@yoga710 ~]$ uname -a
Linux yoga710 4.13.0-rc5-g510c8a899caf #1 SMP PREEMPT Wed Aug 16 13:15:20 EDT 2017 x86_64 GNU/Linux
[parker@yoga710 ~]$ systemctl status iio-sensor-proxy.service 
● iio-sensor-proxy.service - IIO Sensor Proxy service
   Loaded: loaded (/usr/lib/systemd/system/iio-sensor-proxy.service; static; vendor preset: disabled)
   Active: active (running) since Wed 2017-08-16 18:54:05 EDT; 1min 0s ago
 Main PID: 392 (iio-sensor-prox)
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/iio-sensor-proxy.service
           └─392 /usr/bin/iio-sensor-proxy

Aug 16 18:54:05 yoga710 systemd[1]: Starting IIO Sensor Proxy service...
Aug 16 18:54:05 yoga710 systemd[1]: Started IIO Sensor Proxy service.
````

Seems to start fine and without those messages but I still get undefined.

parkerlreed, 2017-08-16:
I can confirm that I still need a suspend/wake cycle to make things work (first time I had actually tried that method)

hadess, 2017-08-16:
> I can confirm that I still need a suspend/wake cycle to make things work (first time I had actually tried that method)

I'm pretty certain you didn't apply the patch properly, or didn't build the kernel properly. Please file a new bug with the debug information requested in the README.

parkerlreed, 2017-08-16:
Well crap. I know what happened. I applied the patch but had to cancel that build. Restarted (PKGBUILD/makepkg) and it rebased the source tree against master, wiping out my patch. Thanks for all the help. I assume everything will go fine after another compile. :)

parkerlreed, 2017-09-28:
@hadess Seems the fix landed in 4.13.3 for me. Thanks again for all the help :)

matheusfillipe, 2020-03-28:
I have this bug on a dell inspiron i14 5481, ubuntu 18.04.4 with kernel 5.3.0-42, monitor-sensor reports "==Has accelerometer (orientation: undefined)". There is no auto rotation.