hadess/iio-sensor-proxy #161: Portrait orientations are upside down on Complex 11t on Ubuntu 17.04

hellslinger, 2017-04-18:
I'm using this obscure little 2-in-1 called the complex 11t, and the rotation worked in 16.10 (for the first time, delightful!), but doesn't work in 17.04. When in portrait mode, the screen is upside down.

Like some other people who have posted issues here, I cannot figure out how to make the matching string "sensor:modalias:...:dmi" line in the relevant 60-sensor.hwdb file. In Ubuntu 17.04, there is no /etc/udev/hwdb.d/60-sensor.hwdb file, so I created one. I don't know if Ubuntu honors this path, so maybe someone can tell me where it goes. I know that I must modify the ACCEL_MOUNT_MATRIX variable in that file once the matching string is correct.

Here is the stanza from udevadm info --export-db that shows the device info for the sensors on this machine:

P: /devices/pci0000:00/INT3432:00/i2c-5/i2c-KIOX000A:00
E: DEVPATH=/devices/pci0000:00/INT3432:00/i2c-5/i2c-KIOX000A:00
E: DRIVER=kxcjk1013
E: ID_VENDOR_FROM_DATABASE=Kionix, Inc.
E: MODALIAS=acpi:KIOX000A:KIOX000A:
E: SUBSYSTEM=i2c
E: USEC_INITIALIZED=2449526

FWIW the module name is actually "kxcjk_1013".

Can someone tell me which file to create for Ubuntu 17.04 and what matching string? I can iterate on the matrix and create a pull request afterward. Thanks.

sarangnemo, 2017-04-22:
I don't know if it's the same on Ubuntu, but on my (Manjaro) system, it's at /usr/lib/udev/hwdb.d, while /etc/udev/hwdb.d exists but is empty!

sarangnemo, 2017-04-22:
To be clear, on my system, /usr/lib/udev/hwdb.d contains profiles for all kinds of other hardware, though I don't have a pre-existing 60-sensor.hwdb file or equivalent information. I'm experimenting with a similar issue on a Surface Pro 3.

RussianNeuroMancer, 2017-04-23:
I trying to apply custom ACCEL_MOUNT_MATRIX on DEXP Ursus 10XW, but have same problem as @hellslinger - variable just don't appear in `sudo udevadm info /dev/iio:device0` output, while /etc/udev/hwdb.d/61-sensor.hwdb is created per [this recommendations](https://github.com/systemd/systemd/blob/master/hwdb/60-sensor.hwdb): 
```
sensor:modalias:acpi:SMO8500*:dmi:*svn*DEXP*:*pn*DEXPOEM*
ACCEL_MOUNT_MATRIX=0, -1, 0; -1, 0, 0; 0, 0, 1
```
There is also no /lib/udev/rules.d/60-sensor.rules in Ubuntu Gnome 17.04, so I tried to place [this file](https://github.com/systemd/systemd/blob/master/hwdb/60-sensor.hwdb) into /lib/udev/rules.d/. Unfortunately this doesn't help, ACCEL_MOUNT_MATRIX doesn't apply, monitor-sensor return incorrect data.

hadess, 2017-04-24:
Did you check whether the version of systemd was new enough, as per README?

RussianNeuroMancer, 2017-04-24:
@hadess Yep, you are right. systemd is 232 which is not new enough. @hellslinger what's your systemd version?

hellslinger, 2017-04-24:
Ok, I have systemd 233 now, but I'd still like to know how to write the modalias because it still doesn't orient correctly.

hellslinger, 2017-04-24:
Alright, this is probably bad syntax, but this works for me now. I'm using latest git versions of systemd and iio-sensor-proxy on Antergos. The AUR made building both packages way easier, though I'm sure it can be done with considerable pain on Ubuntu.
c
sensor:modalias:acpi:*KIOX000A*:*
 ACCEL_MOUNT_MATRIX=-1, 0, 0; 0, 1, 0; 0, 0, 1

Tuxman2, 2017-04-26:
Hello,

I've got the same problem. The screen rotation is not set correctly after login screen for me. The display of my touchscreen is set to Counterclockwise by default (Cinnamon). If I move the screen, this one is inverted in the opposite direction. I followed this topic and here are the informations for my hardware (Medion Akoya):

> P: /devices/platform/80860F41:02/i2c-2/i2c-SMO8500:00
E: DEVPATH=/devices/platform/80860F41:02/i2c-2/i2c-SMO8500:00
E: DRIVER=kxcjk1013
E: ID_VENDOR_FROM_DATABASE=STMicroelectronics
E: MODALIAS=acpi:SMO8500:SMO8500:
E: SUBSYSTEM=i2c
E: USEC_INITIALIZED=7570830

> P: /devices/platform/80860F41:02/i2c-2/i2c-SMO8500:00/iio:device0
N: iio:device0
E: DEVNAME=/dev/iio:device0
E: DEVPATH=/devices/platform/80860F41:02/i2c-2/i2c-SMO8500:00/iio:device0
E: DEVTYPE=iio_device
E: MAJOR=244
E: MINOR=0
E: SUBSYSTEM=iio
E: SYSTEMD_WANTS=iio-sensor-proxy.service
E: TAGS=:systemd:
E: USEC_INITIALIZED=8209717

How to define the "sensor:modalias:...:dmi" line and the ACCEL_MOUNT_MATRIX line for me ?
I'm on Linux Mint 18.1 (based on Ubuntu 16.04) but systemd is 229 version, and for now, systemd 233 version is not yet available. :-(

Can you help me ?

Thanks.

hellslinger, 2017-04-26:
@Tuxman2, you will have to upgrade to systemd 233, which is available only in the Artful (17.10) packages at the moment. You could try to build systemd from source and install, but this is a risky thing to do in Ubuntu.

Tuxman2, 2017-04-27:
@hellslinger: You said it worked on Ubuntu 16.10 but not on Ubuntu 17.04. I used Linux Mint 18.0 and I didn't have this problem. On Linux Mint 18.1, I have it. Very strange. Is it a regression of the version of systemd ? I think it will be more easily for everyone to have a patch to solve this problem than using a quirk. I hope that someone could release a patch soon.

hellslinger, 2017-04-27:
@Tuxman2 correct, in fact, I didn't even realize there was rotation working until I decided to write a gnome extension to provide rotation buttons, and then I picked up my tablet and it rotated automatically.

I suspect that the working rotation was accidental on 16.10 and Mint 18.0, and that in order to accommodate all devices, quirks must be added. This makes perfect sense: each devices mounts its sensors  with different orientations relative to what the default 'correct' orientation is. The quirks system is a good solution since it's just a single line in a config file. 

I've submitted a pull request for systemd here https://github.com/systemd/systemd/pull/5807.

Tuxman2, 2017-04-28:
@hellslinger: Mint 18.0 and Mint 18.1 are based on Ubuntu 16.04. Systemd version for both are 229. Ubuntu 16.10 have a 231 version of Systemd version and Ubuntu 17.04 have a 232 version of Systemd version. I will try the quirk with the 229 version to see what happens. If it doesn't work this will mean that the quirk is only supported on 233 version of systemd. But I think there is probably another possibility or another workaround than using systemd 233 version to solve this probem, no ?

Tuxman2, 2017-04-30:
@All: Well, I have my values for the line "sensor:modalias:...:dmi", but I don't know what values I must define for the ACCEL_MOUNT_MATRIX line. I tried the values given by RussianNeuroMancer for ACCEL_MOUNT_MATRIX line, but as him it doesn't work. The big problem is that we don't have the systemd 233 version on Ubuntu or Linux Mint (it will only be available in the end of this year) and compiling and installing systemd 233 version from source could be a risk like said hellslinger for us. That's why we need another workaround. 

Otherwise, I have a little problem, perhaps you can help me ? I tried to find a solution but for now I haven't solved this problem. My touchscreen is not well recognized. In dmesg error log I've got this message "i2c_hid i2c-FTSC1000:00: hid_descr_cmd failed". It seems there is a problem with the HID descriptor (FTSC1000:00 corresponds to my touchscreen). To make my touchscreen works, I need to remove my i2c_hid module and reload it at start ( I use these commands : modprobe -r i2c_hid and modprobe i2c_hid), otherwise it doesn't work. I would like to know if there is a patch or a workaround to solve this problem because I can't use correctly my touchscreen and this error is a wound for the use of my hardware. Thanks.

hadess, 2017-06-12:
I'm really not sure what I'm supposed to do with the comments about other problems. This isn't a forum, and I can't be expected to unravel all the comments every time you randomly test something different.

> sensor:modalias:acpi:*KIOX000A*:*

This is too narrow, use DMI as well, and when you have something that's specific to your device, please file an issue against systemd with the quirk you used. Bonus points for Pull Requests.

> The big problem is that we don't have the systemd 233 version on Ubuntu or Linux Mint [...]. That's why we need another workaround, of course if this is possible.

The quirk is the work-around. You could ask your distribution to backport the functionality as well. Nothing I can do in iio-sensor-proxy, although you could always hardcode the quirk in the source code even if that's hardly a permanent fix.

closed 2017-06-12 by hadess