hadess/iio-sensor-proxy #165: Asus Q534UX AccelerometerOrientation = 'undefined'

pblanton, 2017-05-10:
I have high hopes for this utility. My Asus Q534UX runs Linux great in all respects except for this issue. I have a cool convertible laptop but can't get it to "convert".

You can see below that I cannot tell teh accelerometer orientation, but the accelerometer is detected.

      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';

Running  monitor-sensor returns this...

Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor

Running "gdbus introspect --system --dest net.hadess.SensorProxy --object-path /net/hadess/SensorProxy" returns the following...

node /net/hadess/SensorProxy {
  interface org.freedesktop.DBus.Properties {
    methods:
      Get(in  s interface_name,
          in  s property_name,
          out v value);
      GetAll(in  s interface_name,
             out a{sv} properties);
      Set(in  s interface_name,
          in  s property_name,
          in  v value);
    signals:
      PropertiesChanged(s interface_name,
                        a{sv} changed_properties,
                        as invalidated_properties);
    properties:
  };
  interface org.freedesktop.DBus.Introspectable {
    methods:
      Introspect(out s xml_data);
    signals:
    properties:
  };
  interface org.freedesktop.DBus.Peer {
    methods:
      Ping();
      GetMachineId(out s machine_uuid);
    signals:
    properties:
  };
  interface net.hadess.SensorProxy {
    methods:
      ClaimAccelerometer();
      ReleaseAccelerometer();
      ClaimLight();
      ReleaseLight();
    signals:
    properties:
      readonly b HasAccelerometer = true;
      readonly s AccelerometerOrientation = 'undefined';
      readonly b HasAmbientLight = false;
      readonly s LightLevelUnit = 'lux';
      readonly d LightLevel = 0.0;
  };
  node Compass {
  };
};

MBus123, 2017-05-16:
Did you try an older version of iio-sensor-proxy? For example the one in the AUR?
On my device (Yoga 900) was an option in the Bios called 'Intel Virtual Technology'. After enabling it, the output was correct.

pblanton, 2017-05-28:
Thanks anyway, but I returned the laptop and bought the HP. The Asus was junk.

hadess, 2017-06-12:
As that's unfixable, closing.

closed 2017-06-12 by hadess

pblanton, 2017-06-12:
Hadess,

That's what I thought. The Asus is "unfixable" junk. Thanks for verifying.

hadess, 2017-06-12:
That's not what I said. I said I can't fix a bug for which the reporter doesn't the hardware anymore. Nothing to do with the quality of the Asus laptop.