hadess/iio-sensor-proxy #167: HP Spectre x360 Kaby Lake (Archlinux) - monitor-sensor only detects data when manually reading iio dev files

Archer89, 2017-05-15:
Hi guys. I'm running iio-sensor-proxy 2.2 on a HP Spectre x360 Kaby Lake laptop with Archlinux (kernel 4.10.13). My kernel is able to successfully detect the IIO sensors

[sergio@chronos devices]$ ls  /sys/bus/iio/devices/
iio:device0  iio:device4  iio:device8  trigger2  trigger6
iio:device1  iio:device5  iio:device9  trigger3  trigger7
iio:device2  iio:device6  trigger0     trigger4  trigger8
iio:device3  iio:device7  trigger1     trigger5  trigger9

And monitor-sensor is able to find them

[sergio@chronos devices]$ monitor-sensor 
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: normal)
=== No ambient light sensor

Nevertheless, it seems monitor-sensor (and Gnome shell) only detects changes in the screen rotation when I'm running this script which continuously tries to read the sensor values:

[sergio@chronos devices]$ cd /sys/bus/iio/devices/
[sergio@chronos devices]$ while true; clear; do { for f in iio\:device*/in_accel*raw; do echo $f`cat $f`; done }; sleep 0.25s; done

iio:device6/in_accel_x_raw-4413
iio:device6/in_accel_y_raw-957076
iio:device6/in_accel_z_raw-208214
iio:device8/in_accel_x_raw-3537
iio:device8/in_accel_y_raw-934787
iio:device8/in_accel_z_raw-203112

Why is not iio-sensor-proxy polling these values by itself or being notified when these values are modified?

detached, 2017-05-16:
Looks like the same problem that I have with https://github.com/hadess/iio-sensor-proxy/issues/164. The Yoga 370 has also a Kaby Lake inside.

JacobMillward, 2017-05-17:
I have this same issue on a Dell Inspiron 13 5378. It also has a Kaby Lake.

Archer89, 2017-06-12:
It seems the issue is fixed after upgrading the system. Still running iio-sensor-proxy 2.2 but now running kernel 4.11.3. It probably was a bug in the kernel but I'm not sure.

closed 2017-06-12 by Archer89

hadess, 2017-06-12:
Kernel bug is still there in 4.12 rc3, so you're just helped by a work-around we put in iio-sensor-proxy.
I've added a section to the README about that.