hadess/iio-sensor-proxy #168: HP x2 210: screen rotation not detected

GreatEmerald, 2017-05-22:
Running monitor-sensor, the first time around it outputs:
```
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: undefined)
=== No ambient light sensor
```
But if I restart `iio-sensor-proxy`, it rather prints:
```
    Waiting for iio-sensor-proxy to appear
+++ iio-sensor-proxy appeared
=== Has accelerometer (orientation: normal)
=== No ambient light sensor
```
However, rotation is still not detected. It stays that way no matter how I move the screen around.

Output of `G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy`:
```
** (process:1877): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device1/in_scale

** (process:1877): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device1/in_offset
** (process:1877): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/trigger1
** (process:1877): DEBUG: Found IIO buffer accelerometer at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device1
** (process:1877): DEBUG: Found device /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device1 of type accelerometer at IIO Buffer accelerometer
** (process:1877): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/trigger1
** (process:1877): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device1/scan_elements/in_accel_y_en
** (process:1877): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device1/scan_elements/in_accel_x_en
** (process:1877): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device1/scan_elements/in_timestamp_en
** (process:1877): DEBUG: Already enabled sensor /sys/devices/pci0000:00/0000:00:0a.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0004/HID-SENSOR-200073.7.auto/iio:device1/scan_elements/in_accel_z_en
** (process:1877): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1877): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1877): DEBUG: Got type for in_timestamp: is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:1877): DEBUG: Got type for in_accel_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1877): DEBUG: Built channel array for in_accel_x: index: 0, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1877): DEBUG: Built channel array for in_accel_y: index: 1, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1877): DEBUG: Built channel array for in_accel_z: index: 2, is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:1877): DEBUG: Built channel array for in_timestamp: index: 3, is signed: 1, bytes: 8, bits_used: 64, shift: 0, mask: 0x18446744073709551615, be: 0
** (process:1877): DEBUG: Handling driver refcounting method 'ClaimLight' for ambient light sensor device
** (process:1877): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:1877): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:1877): DEBUG: No new data available on 'iio:device1'
** (process:1877): DEBUG: No new data available on 'iio:device1'
** (process:1877): DEBUG: No new data available on 'iio:device1'
```

The HWDB and dmesg are attached.
[dmesg.txt](https://github.com/hadess/iio-sensor-proxy/files/1019770/dmesg.txt)
[udevadm.txt](https://github.com/hadess/iio-sensor-proxy/files/1019783/udevadm.txt)

This might be similar to #163 due to related hardware.

alothings, 2017-05-22:
I also have the same issue and get the same message even after restarting the service: 
"=== Has accelerometer (orientation: undefined)"
Using Manjaro on Lenovo Yoga 460

ribizli, 2017-05-22:
on Linux mint I had to revert to kernel 4.4 to get it work again. no luck with 4.10 and 4.11.

GreatEmerald, 2017-05-22:
And I just got it to work by using the linux-sunxi kernel for Cherry Trail:
https://github.com/jwrdegoede/linux-sunxi

Rotation works fine and `monitor-sensor` also reports rotation events. So apparently this is a kernel issue.

hadess, 2017-06-12:
The problem is likely still there, just better hidden. See the "Known problems" section of the README.

treepleks, 2017-07-23:
I have the exact same issue on an HP elitebook x360 1030 G2. The kernel is 4.10.0-28 on Ubuntu 17.04. iio_sensor_proxy is 2.2 and the source contains the 3 seconds sleep kludge, but it does  not work.

treepleks, 2017-07-27:
I installed the last stable kernel release 4.12.3. It works fine now for me. There were some warnings from the kernel on the i915 firmware, so I updated part of it from the Intel website (Ubuntu Zesty).

hadess, 2017-08-16:
From https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#known-problems:

> Every Linux kernel from 4.3 up to version 4.12 had a bug that made
> made iio-sensor-proxy fail to see any events coming from sensors until the 
> sensor was power-cycled (unplugged and replugged, or suspended and resumed).
> 
> The bug was finally fixed in [this commit](https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/commit/?h=fixes-togreg&id=eafad73ed3851707fa6e3124a255fc049ff9545d)
> in the upstream kernel and backported to stable releases. If you experience
> unresponsive sensors, ask your distributor to make sure this patch was 
> applied to the version you're using.

Please reopen if you still have problems with a kernel containing this fix.

closed 2017-08-16 by hadess