hadess/iio-sensor-proxy #169: Asus T300CHI quirks not registering

rolfpal, 2017-06-11:
Hi there, I have a new Asus T300CHI which works perfectly except for screen rotation, it is always 90 clockwise from actual., the sensor and subsystem seems to function just fine, just sideways

I attempted to add new rules at /etc/udev/hwdb.d/61-sensor-local.hwdb (attached) however it does not seem to apply after attempting to apply or reboot etc.  I also attached the output from modalias.

Thank you,

Rolf Paloheimo
[modalias.txt](https://github.com/hadess/iio-sensor-proxy/files/1066678/modalias.txt)
[61-sensor-local.hwdb.txt](https://github.com/hadess/iio-sensor-proxy/files/1066679/61-sensor-local.hwdb.txt)

hadess, 2017-06-12:
Which version of systemd are you using?

rolfpal, 2017-06-12:
Hi There,

I am using 232-25 as reported by synaptic, I am on Debian Stretch.

Thanks,

Rolf

Rolf Paloheimo
Temperate Systems Incorporated
http://temperate.ca

hadess, 2017-06-12:
You need systemd 233 as mentioned in the README. Please file a new issue against systemd if you still run into problems with the updated systemd, as that's where the quirk would ultimately live.

closed 2017-06-12 by hadess

rolfpal, 2017-06-12:
Thanks for your help
Rolf

Rolf Paloheimo
Temperate Systems Incorporated
http://temperate.ca

rolfpal, 2017-06-23:
I changed to Ubuntu 17.10 daily, and edited
/etc/udev/hwdb.d/61-sensor-local.hwdb, which works.

The file is attached.

Cheerio,

Rolf

Rolf Paloheimo
Temperate Systems Incorporated
http://temperate.ca