hadess/iio-sensor-proxy #171: Lenovo IdeaPad Yoga 13 Rotates off by 90 degrees

sniper, 2017-07-06:
The laptop is listed as tested, but the rotation is always 90 degrees clockwise of what it should be (it thinks the left side of the screen is the "bottom"). 
Here is the output of `udevadm info --export-db`: [out.txt](https://github.com/hadess/iio-sensor-proxy/files/1127289/out.txt)

hadess, 2017-07-06:
Please provide a quirk to systemd's hwdb, after checking your version of systemd is new enough.

See https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#installation and https://github.com/hadess/iio-sensor-proxy/blob/master/README.md#accelerometer-orientation for more info.

closed 2017-07-06 by hadess

offthewoll, 2017-09-11:
I have the ThinkPad Yoga 11e 3rd Gen and have an identical issue.

I'm on version 233 of systemd and read the above links, but still don't understand what I should do.