hadess/iio-sensor-proxy #172: Light level fixed in ASUS UX330UAK

rogervn, 2017-07-11:
Hello, I've installed the AUR iio-sensor-proxy in arch with kernel 4.11.9-1-ARCH, but after running monitor-sensor and debugging with G_MESSAGES_DEBUG_ALL, it seems that the light level keeps stuck on 400.000 and doesn't change even if I turn off the lights.

G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy output:

[iio-sensor-proxy.txt](https://github.com/hadess/iio-sensor-proxy/files/1137587/iio-sensor-proxy.txt)

hadess, 2017-07-11:
You can try and verify whether the light level changes by using "cat" as root in the directory mentioned. If that doesn't change either, then it's a problem with either the device or its driver.

rogervn, 2017-07-12:
cat at /sys/devices/LNXSYSTM\:00/LNXSYBUS\:00/ACPI0008\:00/iio\:device0/in_illuminance_raw  does always return 400. What can I do to check the driver? It works perfectly in Windows.

lsmod output:

[code]
lsmod | grep asus
asus_nb_wmi            24576  0
asus_wmi               24576  1 asus_nb_wmi
sparse_keymap          16384  1 asus_wmi
wmi                    20480  1 asus_wmi
rfkill                 20480  8 asus_wmi,bluetooth,cfg80211
video                  36864  2 asus_wmi,i915
asus_wireless          16384  0
led_class              16384  4 iwlmvm,asus_wmi,input_leds,asus_wireless
i8042                  24576  1 asus_nb_wmi
[/code]

hadess, 2017-07-12:
I'm afraid I can't do anything about it then. Please report the bug to the driver developer(s).

closed 2017-07-12 by hadess