hadess/iio-sensor-proxy #173: HP Envy x360 2 in 1... not working

9acca9, 2017-07-16:
Hi to all.
Can somebody give me a hand? 
i get this https://imgur.com/ohyUCg1 in boot.
the install iio-sensor-proxy but its not working...

Linux version 4.11.9-1-ARCH

jul 15 23:17:13 WindowsNew kernel: lis3lv02d: unknown sensor type 0x0
jul 15 23:17:13 WindowsNew kernel: hp_accel: probe of HPQ6007:00 failed with error -22
jul 15 23:17:13 WindowsNew kernel: ACPI: Invalid passive threshold
jul 15 23:17:13 WindowsNew kernel: (NULL device *): hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().

[xxxxx@WindowsNew ~]$ systemctl start iio-sensor-proxy.service
[xxxxx@WindowsNew ~]$ systemctl status iio-sensor-proxy.service
● iio-sensor-proxy.service - IIO Sensor Proxy service
   Loaded: loaded (/usr/lib/systemd/system/iio-sensor-proxy.service; static; ven
   Active: inactive (dead)

(i dont speak english)
Thanks to all.

9acca9, 2017-07-16:
Well... i restart the laptop and is working, however i still get exactly this 
lis3lv02d: unknow sensor type

and, by the way every time i rotate the screen the display turn off, (it works, but turns off every time i rotate the screen, there is a way to fix this????)

Thanks to all.

hadess, 2017-08-16:
That warning comes from a kernel module, not from iio-sensor-proxy. Please report it to the developers of the kernel driver, or your distribution if you don't know how to do that.

The screen flickering is a problem in the desktop stack. Again, iio-sensor-proxy doesn't not make that change, it's a problem with the desktop environment you use, and that's where the bug report should be made (or with your distribution which will report it upstream).

closed 2017-08-16 by hadess