hadess/iio-sensor-proxy #174: Maxing Out Processor with Fedora on HP Envy Laptop

spitofland, 2017-07-16:
I'm running Fedora 25 (Linux kernel 4.11.8-200.fc25.x86_64) on an HP ENVY dv4 laptop. I found that the iio-sensor-proxy process was maxing out one of my cores. When I stopped the iio-sensor-proxy service, my CPU usage levels dropped to expected idle levels and the CPUs began throttling to conserve power.

I'm not sure why this process would consume 100% of a core's time during what should be idle conditions. I can only think of two sensors that my laptop might have, a light sensor and a lid-close sensor. Closing the lid of my laptop still triggered sleep mode when the service was stopped, so I will probably be fine with just turning the service off, but this issue could cause a reduction in battery life if the service is running.

hadess, 2017-07-16:
There's information in the README on how to gather debug information.

spitofland, 2017-07-16:
Thank you.
I ran the debug command in the README and received the following output.

> ** (process:9321): DEBUG: Found input accel at /sys/devices/platform/lis3lv02d/input/input20/event13
** (process:9321): DEBUG: Found device /sys/devices/platform/lis3lv02d/input/input20/event13 of type accelerometer at Input accelerometer
** (process:9321): DEBUG: Accel read from input on '(null)': -180, 0, 1152 (scale 0.038320)
** (process:9321): DEBUG: Accel sent by driver (quirk applied): -180, 0, 1152 (scale: 0.038320)
** (process:9321): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:9321): DEBUG: Handling driver refcounting method 'ClaimAccelerometer' for accelerometer device
** (process:9321): DEBUG: Accel read from input on '(null)': -180, 0, 1152 (scale 0.038320)
** (process:9321): DEBUG: Accel sent by driver (quirk applied): -180, 0, 1152 (scale: 0.038320)
** (process:9321): DEBUG: Accel read from input on '(null)': -180, 0, 1152 (scale 0.038320)
** (process:9321): DEBUG: Accel sent by driver (quirk applied): -180, 0, 1152 (scale: 0.038320)

The last two lines get repeated about every 10 seconds. The process `iio-sensor-proxy` almost immediately maxed out one of my cores when it started.

hadess, 2017-07-16:
Looks like an older version of iio-sensor-proxy. Try upgrading it.

I also doubt that iio-sensor-proxy is actually using 100% of the CPU if it's waking up every 10 seconds, but it's possible that the device is sending tons of updates. You'll want to install `evtest` and see whether you get a constant stream of events, like 10s of thousands per second, because that's pretty much the only way it would eat this much CPU.

spitofland, 2017-07-17:
I'll try updating. When I tried evtest, it did not show any accelerometer events and the other events only occurred when an input was given (much less than thousands of times a second, even during trackpad inputs). When I made sure iio-sensor-proxy was running in debug before calling evtest, the evtest output did not change, but the iio-sensor-proxy debug output started repeating the last two lines a little more frequently (once every 3-4 seconds).

hadess, 2020-01-09:
Moved to https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/issues/174

closed 2020-01-09 by hadess