hadess/iio-sensor-proxy #180: iio-sensor-proxy attempts to read from non existent files.

hoskeri, 2017-08-04:
Running debian kernel 4.12.0, and iio-sensor-proxy 2.2.

** (process:22035): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-14/i2c-NPCE0388:00/0018:0416:C300.0001/HID-SENSOR-200073.1.auto/iio:device0/in_scale

** (process:22035): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-14/i2c-NPCE0388:00/0018:0416:C300.0001/HID-SENSOR-200073.1.auto/iio:device0/in_offset

M0Rf30, 2017-08-05:
Same problem here. Anyone that could symlink these
in_scale -> in_accel_scale
in_offset -> in_accel_offset

with udev rules?

hadess, 2017-08-16:
The warnings were red herrings, they shouldn't have been printed. This is fixed in the commit mentioned above.

As for the `watch` command that you needed, it's a fallout of an IIO subsystem bug, see the "Known problems" section of the README.

closed 2017-08-16 by hadess

M0Rf30, 2017-08-16:
@hadess 
Anyway you could add Lenovo Yoga 520 14'' to tested devices
Cheers

hadess, 2017-08-16:
Please file a new bug once you've actually tested it. Or don't. I should just remove this section as we support pretty the majority of devices nowadays.

commit d0ced556 (2017-08-16): iio-buffer-utils: Don't throw warnings on ENOENT

As we don't return when the file doesn't exist, don't print a warning
either, as this would pollute logs and throw people off when debugging
real problems.

Closes: #180