hadess/iio-sensor-proxy #182: Lenovo Yoga 520 (80x80 variant) - undefined orientation

M0Rf30, 2017-09-02:
After having updated the kernel to 4.12.10 (containing the famous [fix](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=v4.12.10&id=fc7957b6cdd787e70923dfce250f0564590fe922) referenced in README),
I've updated iio-sensor-proxy to latest commit, libudev to 232 and here I post the log with DEBUG variable enabled.

https://gist.github.com/M0Rf30/8ed0318ebaa87c790579ab1dfc5d6360

monitor-sensor gives "undefined"
after sending this command:
` watch -n 0.1 cat '/sys/bus/iio/devices/iio:device'*/*raw*`
it works like a charm and the relative entry in journal become

`set 02 13:04:48 unknown iio-sensor-prox[414]: process_scan_1.....`

Stopping the watch process it returns to

`set 02 13:04:50 unknown iio-sensor-prox[414]: No new data available on 'iio:device0'
`

hadess, 2017-09-02:
Any i2c errors in the kernel logs? Does a suspend/resume work? Are you certain that you're testing the new kernel and the new iio-sensor-proxy?

M0Rf30, 2017-09-02:
Suspend / resume seems to work. Before the 4.12.10 kernel, watch process didn't work after resume. I've entered "dmesg | grep i2c" and no errors.

Here dmesg:
https://gist.github.com/M0Rf30/ee393fb81b6b6eb9389781e284629258

hadess, 2017-09-04:
I have no idea then, you'll need to contact the upstream kernel maintainer for this subsystem to get this sorted out.

M0Rf30, 2017-09-08:
Bug submitted on Kernel Bugzilla
https://bugzilla.kernel.org/show_bug.cgi?id=196871

dheater, 2017-10-27:
Same here with Lenovo Flex-3. Ubuntu 17.10 with kernel 4.13.0-16.

Orientation reports `undefined` until I run 
```
watch -n 0.1 cat '/sys/bus/iio/devices/iio:device'*/*raw*
```

reneluria, 2017-11-03:
Same here with Lenovo Yoga 720 13IKB ubuntu 17.10 kernel 4.13.0-16 or 4.13.11-041311.2
iio-sensor-proxy 2.2-1 or 2.4
`orientation: undefined` 
until I poll
`/sys/bus/iio/devices/iio:device'*/in_accel_*_raw`

M0Rf30, 2018-01-02:
I don't know how, but for me it's fixed. So I close the issue. Thanks hadess

closed 2018-01-02 by M0Rf30