hadess/iio-sensor-proxy #183: README.MD: Bad commit reference

brokkr, 2017-09-05:
I wanted to check up on how the kernel bug fix was progressing (hopefully) towards release. However, the link in the readme now (unlike previously) takes me to a page saying 'bad commit reference'.

hadess, 2017-09-05:
Which link? The link in the README.md works fine here.

hadess, 2017-09-05:
That's the link in the README that works for me:
https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/commit/?h=fixes-togreg&id=eafad73ed3851707fa6e3124a255fc049ff9545d

brokkr, 2017-09-07:
Sorry, yes, that's the one I was referring to. When you say "works for me" do you mean that you do not get an empty page only saying "bad commit reference"? I tried different browsers on different machines from different networks and that was all I was getting.

OK, I just tried screenshotting it using a screenshot service (https://www.url2png.com/#testdrive) to show it and the service clearly sees the actual commit. My bad.

hadess, 2017-09-07:
It works means that I can see the commit just fine. You might want to check your system for other errors (maybe a VPN, proxy, or ad blocker is causing problems).

I'll close this now, thanks.

closed 2017-09-07 by hadess