hadess/iio-sensor-proxy #184: Dell Inspiron 7378 monit-sensor does not detect rotation

sjauld, 2017-09-06:
I have installed kernel 4.13.0-041300-generic.
When I run monit-sensor, nothing seems to happen

```
$  sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:5149): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device0
** (process:5149): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:5149): DEBUG: Found magn_3d at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.16.auto/iio:device3
** (process:5149): DEBUG: Found device /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.16.auto/iio:device3 of type compass at IIO Buffer Compass
** (process:5149): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/trigger0
** (process:5149): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5149): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0

** (process:5149): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device0/in_scale

** (process:5149): WARNING **: Problem reading scan element information: /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device0
** (process:5149): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200083.16.auto/trigger3
** (process:5149): DEBUG: Got type for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5149): DEBUG: Got type for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5149): DEBUG: Got type for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5149): DEBUG: Got type for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5149): DEBUG: Built channel array for in_magn_z: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5149): DEBUG: Built channel array for in_magn_y: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5149): DEBUG: Built channel array for in_rot_from_north_magnetic_tilt_comp: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:5149): DEBUG: Built channel array for in_magn_x: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0

```

hadess, 2017-09-06:
I've made a small change to iio-sensor-proxy, so we can try and see where the problem is:
https://github.com/hadess/iio-sensor-proxy/commit/8e994daa3bedab19b146e3eab3481b93f461e314

Could you please try and apply it to iio-sensor-proxy and gather the debug again?
Do you see any error messages in the kernel logs when this happens?

sjauld, 2017-09-07:
Revised error message:
```
** (process:28076): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device0/in_scale: No such file or directory
```

Also,
```
$ sudo dmesg
[10746.571623] intel-vbtn INT33D6:00: unknown event index 0xcb
[10746.571837] intel-vbtn INT33D6:00: unknown event index 0xcc
[10746.594825] dell_wmi: Unknown key with type 0x0000 and code 0xe070 pressed
[10749.286737] intel-vbtn INT33D6:00: unknown event index 0xca
[10749.286773] intel-vbtn INT33D6:00: unknown event index 0xcd
[10749.305027] dell_wmi: Unknown key with type 0x0000 and code 0xe070 pressed
```
Interestingly, no keys seem to work once I rotate the monitor.

hadess, 2017-09-07:
> ** (process:28076): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:13.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001F:8086:22D8.0002/HID-SENSOR-200073.11.auto/iio:device0/in_scale: No such file or directory

Did you apply the patch correctly? The message is not supposed to throw any warnings when the file is missing, and it would just skip over it.

This message is not fatal either.

Huiko, 2017-09-09:
Hello,
Have the same issue on Lenovo Yoga 910-13IKB. My output looks like this
```
sudo G_MESSAGES_DEBUG=all /usr/sbin/iio-sensor-proxy
** (process:18556): DEBUG: Found als at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-7/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1
** (process:18556): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-7/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/iio:device1 of type ambient light sensor at IIO Buffer Light sensor
** (process:18556): DEBUG: Found accel_3d at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-7/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0
** (process:18556): DEBUG: Found device /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-7/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0 of type accelerometer at IIO Buffer accelerometer
** (process:18556): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-7/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/trigger0
** (process:18556): DEBUG: Got type for in_accel_y: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0
** (process:18556): DEBUG: Got type for in_accel_x: is signed: 1, bytes: 4, bits_used: 16, shift: 0, mask: 0x65535, be: 0

** (process:18556): WARNING **: Failed to read float from /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-7/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0/in_scale

** (process:18556): WARNING **: Problem reading scan element information: /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-7/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200073.2.auto/iio:device0
** (process:18556): DEBUG: Found associated trigger at /sys/devices/pci0000:00/0000:00:15.3/i2c_designware.2/i2c-7/i2c-ITE8186:00/0018:048D:8186.0003/HID-SENSOR-200041.3.auto/trigger1
** (process:18556): DEBUG: Got type for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
** (process:18556): DEBUG: Built channel array for in_intensity_both: is signed: 1, bytes: 4, bits_used: 32, shift: 0, mask: 0x0, be: 0
```

And sorry, don't want to look dumb, but I'm totaly new to github and somewhat new to linuxing. How do I apply your little patch?

jonasBoss, 2017-09-10:
Hello,
I have a similar issue on Lenovo Yoga 510-14IKB.

```
$ uname -a
Linux yoga 4.13.0-mainline #1 SMP PREEMPT Sat Sep 9 23:47:01 CEST 2017 x86_64 GNU/Linux
```
I applied the patch 8e994da to https://aur.archlinux.org/packages/iio-sensor-proxy (version 2.2-3) and got the following output:

```
** (process:4502): WARNING **: Failed to read float from 
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-NPCE0388:00/0018:0416:C300.0001/HID-SENSOR-200073.1.auto/iio:device0/in_scale: 
No such file or directory

** (process:4502): WARNING **: Failed to read float from 
/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-NPCE0388:00/0018:0416:C300.0001/HID-SENSOR-200073.1.auto/iio:device0/in_offset: 
No such file or directory
```
[log.txt](https://github.com/hadess/iio-sensor-proxy/files/1290446/log.txt)

I did look up the directory and found the following:

```
jonas@yoga:/sys/devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-NPCE0388:00/0018:0416:C300.0001/HID-SENSOR-200073.1.auto/iio:device0$ ls -la
insgesamt 0
drwxr-xr-x 6 root root    0 10. Sep 11:56 .
drwxr-xr-x 5 root root    0 10. Sep 11:56 ..
drwxr-xr-x 2 root root    0 10. Sep 11:56 buffer
-rw-r--r-- 1 root root 4096 10. Sep 12:19 current_timestamp_clock
-r--r--r-- 1 root root 4096 10. Sep 12:19 dev
-rw-r--r-- 1 root root 4096 10. Sep 12:19 in_accel_hysteresis
-rw-r--r-- 1 root root 4096 10. Sep 12:19 in_accel_offset
-rw-r--r-- 1 root root 4096 10. Sep 12:19 in_accel_sampling_frequency
-rw-r--r-- 1 root root 4096 10. Sep 12:19 in_accel_scale
-rw-r--r-- 1 root root 4096 10. Sep 12:19 in_accel_x_raw
-rw-r--r-- 1 root root 4096 10. Sep 12:19 in_accel_y_raw
-rw-r--r-- 1 root root 4096 10. Sep 12:19 in_accel_z_raw
-r--r--r-- 1 root root 4096 10. Sep 12:19 name
drwxr-xr-x 2 root root    0 10. Sep 12:19 power
drwxr-xr-x 2 root root    0 10. Sep 11:56 scan_elements
lrwxrwxrwx 1 root root    0 10. Sep 11:56 subsystem -> 
../../../../../../../../../bus/iio
drwxr-xr-x 2 root root    0 10. Sep 11:56 trigger
-rw-r--r-- 1 root root 4096 10. Sep 12:19 uevent
```

there seem to be a mismatch between the error Message `iio:device0/in_scale` and the actual file I found `iio:device0/in_accel_scale` 

however this might be related to bug [#182](https://github.com/hadess/iio-sensor-proxy/issues/182) 

`$ watch -n 0.1 cat '/sys/bus/iio/devices/iio:device'*/*raw*`
rendered my Gnome-shell rotation functional

sjauld, 2017-09-11:
> Did you apply the patch correctly? The message is not supposed to throw any warnings when the file is missing, and it would just skip over it.

Ah yes, I'm running version 1.1 which doesn't have that `if (ret != -ENOENT)` line.

I'm going to go away and upgrade Ubuntu to 17.04 then try again

sjauld, 2017-09-11:
Looks like the upgrade did the trick! Sorry about the red herring...

hadess, 2017-09-11:
Alright then. Not sure what the problem was but I can always think to myself that it was something I did in the past ;)

Thanks for the notice!

closed 2017-09-11 by hadess