hadess/iio-sensor-proxy #186: QUESTION: How can I add touchpad disabling to the rotation event?

anixon604, 2017-09-26:
I have debian 9 / gnome with kernel 4.9 on Lenovo X1 Yoga. iio-sensor works great and I don't see an issue yet. The only thing I need now is to make my touchpad click disable when I'm in tablet mode. Can someone give me some direction or guidance on how to do this?

Thank you.

Hnasar, 2017-09-27:
You'd need to watch for the Dbus `SensorProxy` name and then get the dbus property for the `AccelerometerOrientation`. See `mutter/src/backends/meta-orientation-manager.c` or `gnome-settings-daemon/plugins/power/gsd-power-manager.c` for examples.

After you read the values from the accelerometer then you'll need to flip the `gsettings` settings responsible for enabling/disabling touchpad (not sure where exactly, probably something like `org/gnome/desktop/peripherals/touchpad`?)

hadess, 2017-10-04:
API documentation for iio-sensor-proxy is here:
https://developer.gnome.org/iio-sensor-proxy/2.3/

closed 2017-10-04 by hadess