hadess/iio-sensor-proxy #188: Support BMA2x2 accelerometer

Ilia1, 2017-10-05:
command output monitor-sensor contains nothing.

hadess, 2017-10-05:
Please provide the debug output of iio-sensor-proxy as per README.

Ilia1, 2017-10-06:
** (process:9488): DEBUG: Could not find any supported sensors

hadess, 2017-10-06:
What's the output of `udevadm info --export-db` on this system? Is there actually a driver for this device? What system is this in?

Ilia1, 2017-10-07:
[`udevadm info --export-db`](https://pastebin.com/9k5Yx7Ni)
System Linux Mint 18.2, kernel 4.13.1, version iio-sensor-proxy 1.1-1ubuntu1

hadess, 2017-10-07:
You should test at least version 2.0, which has a number of detection fixes. This version is 2 years old, which is very old for such a young project.

Ilia1, 2017-10-07:
thank you very much, updating to the latest version helped. now tell me, which script or program to choose to change the screen orientation?

hadess, 2017-10-07:
> thank you very much, updating to the latest version helped.

Great, I'll close this bug then.

> now tell me, which script or program to choose to change the screen orientation?

It's up to the desktop environment to provide that. I implemented it in GNOME, and it's been there for a while. Anything else, you can point the desktop developers towards this API documentation for them to use:
https://developer.gnome.org/iio-sensor-proxy/

closed 2017-10-07 by hadess