hadess/iio-sensor-proxy #189: PS3 gamepad blanks my monitor when I move it

bochecha, 2017-10-05:
I have a PS3 bluetooth gamepad.

I plugged it with USB to my laptop in order to charge it, and at some point it fell on the floor. My external monitor just blacked out for a few seconds, then came back.

I can reproduce it, moving the gamepad.

@hadess made me try [the latest iio-sensor-proxy](https://bodhi.fedoraproject.org/updates/libgudev-232-1.fc26%20iio-sensor-proxy-2.3-1.fc26) to see if it helped, but it didn't. (I did reboot after updating those 2 packages)

If I enable debug messages for iio-sensor-proxy, I continuously get the following two lines, over and over again in the journal, even if I'm not touching anything at all:

```
Oct 05 19:27:48 mathieus-laptop iio-sensor-prox[7635]: Accel read from input on '(null)': 10, 66, 6 (scale 0.038320)
Oct 05 19:27:48 mathieus-laptop iio-sensor-prox[7635]: Accel sent by driver (quirk applied): 10, 66, 6 (scale: 0.038320)
```

Then when I picked up the gamepad and moved it, I got:

```
Oct 05 19:28:38 mathieus-laptop iio-sensor-prox[7635]: Accel read from input on '(null)': 65, 72, 42 (scale 0.038320)
Oct 05 19:28:38 mathieus-laptop iio-sensor-prox[7635]: Accel sent by driver (quirk applied): 65, 72, 42 (scale: 0.038320)
Oct 05 19:28:38 mathieus-laptop iio-sensor-prox[7635]: Emitted orientation changed: from bottom-up to left-up
Oct 05 19:28:39 mathieus-laptop iio-sensor-prox[7635]: Accel read from input on '(null)': 75, 4, -21 (scale 0.038320)
Oct 05 19:28:39 mathieus-laptop iio-sensor-prox[7635]: Accel sent by driver (quirk applied): 75, 4, -21 (scale: 0.038320)
Oct 05 19:28:40 mathieus-laptop iio-sensor-prox[7635]: Accel read from input on '(null)': 27, 63, 8 (scale 0.038320)
Oct 05 19:28:40 mathieus-laptop iio-sensor-prox[7635]: Accel sent by driver (quirk applied): 27, 63, 8 (scale: 0.038320)
Oct 05 19:28:40 mathieus-laptop iio-sensor-prox[7635]: Emitted orientation changed: from left-up to bottom-up
```

The last two lines now repeat over and over.

Here is the output of `sudo udevadm info --export-db`:

[udevadm-info-export-db.txt](https://github.com/hadess/iio-sensor-proxy/files/1360288/udevadm-info-export-db.txt)

hadess, 2017-10-05:
Looks like a race, I don't think that the joypad is setup by the time we check the accelerometer's siblings.

hadess, 2017-10-09:
> I plugged it with USB to my laptop in order to charge it, and at some point it fell on the floor. My external monitor just blacked out for a few seconds, then came back.

Looks like it doesn't send any events until I press the "P" button. So that's the part I was missing to try and reproduce the problem (eg. have events be sent).

But the newer (2.3) version of iio-sensor-proxy correctly ignores the motion sensor as the joypad is there. If you can, can you please add debug in `is_part_of_joypad` and overwrite the system binary:
https://github.com/hadess/iio-sensor-proxy/blob/master/src/drv-input-accel.c#L83
and tell me where it fails?

iio-sensor-proxy is supposed to exit as soon as it finds that the accelerometer has a joypad as a sibling.

hadess, 2017-10-24:
It apparently doesn't fail with the same version as tested, so closing.

closed 2017-10-24 by hadess